NVlabs/curobo #220: python3 -m pytest . error

ThisIsBohao, 2024-04-12:
If it’s not a bug, please use discussions: https://github.com/NVlabs/curobo/discussions

Please provide the below information in addition to your issue:

1. cuRobo installation mode: python with conda, without docker
2. python version: 3.11

------
Issue Details:

I tried to setup a conda environment with only pytorch:
```
name: curobo-env
channels:
  - conda-forge
  - defaults
dependencies:
  - python=3.11
  - numpy
  - pytorch>=1.10
  - tensorboard=2.16.2
  - pip=24.0
  - pip:
    - "ray==2.10.0"
```

I successfully installed curobo by pip install -e . --no-build-isolation. 
Then I run python3 -m pytest . and get the following error:
```
platform linux -- Python 3.11.8, pytest-8.1.1, pluggy-1.4.0
configfile: pyproject.toml
plugins: anyio-4.3.0
collected 3 items / 36 errors 

tests/cost_test.py:15: in <module>
    from curobo.geom.sdf.world import WorldCollisionConfig, WorldPrimitiveCollision
src/curobo/geom/sdf/world.py:22: in <module>
    from curobo.geom.types import Cuboid, Mesh, Obstacle, VoxelGrid, WorldConfig, batch_tensor_cube
src/curobo/geom/types.py:38: in <module>
    @dataclass
../../anaconda3/envs/curobo-env/lib/python3.11/dataclasses.py:1230: in dataclass
    return wrap(cls)
../../anaconda3/envs/curobo-env/lib/python3.11/dataclasses.py:1220: in wrap
    return _process_class(cls, init, repr, eq, order, unsafe_hash,
../../anaconda3/envs/curobo-env/lib/python3.11/dataclasses.py:958: in _process_class
    cls_fields.append(_get_field(cls, name, type, kw_only))
../../anaconda3/envs/curobo-env/lib/python3.11/dataclasses.py:815: in _get_field
    raise ValueError(f'mutable default {type(f.default)} for field '
E   ValueError: mutable default <class 'curobo.geom.types.Material'> for field material is not allowed: use default_factory
```

This happens for literally every script in the test folder. 
I wonder if you have any idea what this issue might be.
I would appreciate it very much if you could help me with this.

balakumar-s, 2024-04-12:
We are working on supporting python 3.11. Until then, can you use python 3.10 or lower?

Cfather, 2024-04-15:
Hi, I tried python 3.10 and the code works fine. Thanks for the help!

ThisIsBohao, 2024-04-15:
Hi, I tried python 3.10 and the code works fine. Thanks for the help!

closed 2024-04-15 by ThisIsBohao