NVlabs/curobo #221: Runtime Error: Constrained Motion Planning

maximiliangilles, 2024-04-15:
Hi,

thanks for your great work! I am trying to run _constrained_motion_planning_.

My example code looks like this:

  ```
self.constrain_grasp_approach = constrain_grasp_approach
  self.tensor_args = TensorDeviceType(device=torch.device("cuda:0"))
  interpolation_dt = 0.02
  collision_activation_distance = 0.02  # meters
  # create
  robot_config = load_yaml("/home/curobo/src/curobo/content/configs/robot/franka.yml")
  cuboids = [
      Cuboid(name="bin1", pose=[0.58, 0.0, 0.035, 1., 0., 0., 0.], dims=[0.01, 0.4, 0.07]),  #x,y,z, qx, qy, qz, qw
      Cuboid(name="bin2", pose=[0.28, 0.0, 0.035, 1., 0., 0., 0.], dims=[0.01, 0.4, 0.07]),  #x,y,z, qx, qy, qz, qw
      Cuboid(name="bin3", pose=[0.45, 0.2, 0.035, 1., 0., 0., 0.], dims=[0.3, 0.01, 0.07]),  #x,y,z, qx, qy, qz, qw
      Cuboid(name="bin4", pose=[0.45, -0.2, 0.035, 1., 0., 0., 0.], dims=[0.3, 0.01, 0.07]),  #x,y,z, qx, qy, qz, qw
      # Cuboid(name="bin", pose=[0.45, 0.0, 0.035, 1., 0., 0., 0.], dims=[0.3, 0.4, 0.07]),  #x,y,z, qx, qy, qz, qw
      Cuboid(name="table", pose=[0.0, 0.0, -0.05, 1., 0., 0., 0.], dims=[5.0, 5.0, 0.1]),  #x,y,z, qx, qy, qz, qw
      Cuboid(name="column1", pose=[0.84, 0.36, 0.5, 1., 0., 0., 0.], dims=[0.08, 0.08, 1.0]),
      Cuboid(name="column2", pose=[0.84, -0.36, 0.5, 1., 0., 0., 0.], dims=[0.08, 0.08, 1.0]),
      Cuboid(name="column3", pose=[-0.26, 0.36, 0.5, 1., 0., 0., 0.], dims=[0.08, 0.08, 1.0]),
      Cuboid(name="column4", pose=[-0.26, -0.36, 0.5, 1., 0., 0., 0.], dims=[0.08, 0.08, 1.0])]  #x,y,z, qx, qy, qz, qw
  
  world_config = WorldConfig(cuboid=cuboids)
  # create motion gen with a cuboid cache to be able to load obstacles later:
  motion_gen_cfg = MotionGenConfig.load_from_robot_config(
      robot_config,
      world_config,
      self.tensor_args,
      trajopt_tsteps=34, # 34 
      interpolation_steps=5000,
      num_ik_seeds=50,
      num_trajopt_seeds=6,
      grad_trajopt_iters=500,
      trajopt_dt=0.5,
      interpolation_dt=interpolation_dt,
      evaluate_interpolated_trajectory=True,
      js_trajopt_dt=0.5,
      js_trajopt_tsteps=34,  # 34
      collision_activation_distance=collision_activation_distance)
  self.motion_gen = MotionGen(motion_gen_cfg)
  self.motion_gen.warmup()
  if self.constrain_grasp_approach:
      pose_metric = PoseCostMetric.create_grasp_approach_metric(
          offset_position=0.05, tstep_fraction=0.2, linear_axis=2
      )
      self.plan_config = MotionGenPlanConfig(
          enable_graph=False,
          max_attempts=10,
          enable_graph_attempt=None,
          enable_finetune_trajopt=True,
          partial_ik_opt=False,
          parallel_finetune=False,
          pose_cost_metric=pose_metric,
      )
  result = self.motion_gen.plan_single(q_start, goal_pose, self.plan_config.clone())
```

I get the follwing error:
```
tensor([ 250., 5000.,   20.,   20.], device='cuda:0')
Updating offset waypoint is only possible after initializing motion gen run motion_gen.warmup() before adding offset_waypoint
NoneType: None
Stack (most recent call last):
  File "/usr/lib/python3.10/threading.py", line 973, in _bootstrap
    self._bootstrap_inner()
  File "/usr/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
    self.run()
[... 66 lines cut ...]
  File "/usr/local/lib/python3.10/dist-packages/curobo/wrap/reacher/motion_gen.py", line 1322, in plan_single
    result = self._plan_attempts(
  File "/usr/local/lib/python3.10/dist-packages/curobo/wrap/reacher/motion_gen.py", line 1097, in _plan_attempts
    valid_query = self.update_pose_cost_metric(
  File "/usr/local/lib/python3.10/dist-packages/curobo/wrap/reacher/motion_gen.py", line 2535, in update_pose_cost_metric
    [
  File "/usr/local/lib/python3.10/dist-packages/curobo/wrap/reacher/motion_gen.py", line 2536, in <listcomp>
    rollout.update_pose_cost_metric(metric)
  File "/usr/local/lib/python3.10/dist-packages/curobo/rollout/arm_reacher.py", line 467, in update_pose_cost_metric
    [
  File "/usr/local/lib/python3.10/dist-packages/curobo/rollout/arm_reacher.py", line 468, in <listcomp>
    x.update_offset_waypoint(
  File "/usr/local/lib/python3.10/dist-packages/curobo/rollout/cost/pose_cost.py", line 207, in update_offset_waypoint
    log_error(
  File "/usr/local/lib/python3.10/dist-packages/curobo/util/logger.py", line 45, in log_error
    raise
RuntimeError: No active exception to reraise
```
I already tried to set` use_cuda_graph=False` or comment out `motion.gen_warmup() `(see https://github.com/NVlabs/curobo/issues/219). Both possbile fix did not work and throw the same error message.

System:
1. cuRobo installation mode: docker python
2. python version: 3.10.12

Thank you very much!

balakumar-s, 2024-04-15:
It looks like warmup failed. Can you warmup with an empty world and then add your world? 

You can do that by:
1. passing `collision_cache = {"obb": 10}` to `MotionGenConfig.load_from_robot_config`
2. once MotionGen instance is created, call `motion_gen.clear_world_cache()`
3. then run motion_gen.warmup()
4. load your world with `motion_gen.update_world(world_config)`

maximiliangilles, 2024-04-16:
Hi @balakumar-s ,

thanks for the fast reply. I tried your proposed answer, however I still get the same error message. I attached my code for you to verify. Do you see any mistakes?
```
self.tensor_args = TensorDeviceType(device=torch.device("cuda:0"))

interpolation_dt = 0.02
collision_activation_distance = 0.02  # meters

# create
robot_config = load_yaml("/home/curobo/src/curobo/content/configs/robot/franka.yml")

cuboids = [
    Cuboid(name="bin1", pose=[0.58, 0.0, 0.035, 1., 0., 0., 0.], dims=[0.01, 0.4, 0.07]),  #x,y,z, qx, qy, qz, qw
    Cuboid(name="bin2", pose=[0.295, 0.0, 0.035, 1., 0., 0., 0.], dims=[0.01, 0.4, 0.07]),  #x,y,z, qx, qy, qz, qw
    Cuboid(name="bin3", pose=[0.4375, 0.19, 0.035, 1., 0., 0., 0.], dims=[0.3, 0.01, 0.07]),  #x,y,z, qx, qy, qz, qw
    Cuboid(name="bin4", pose=[0.4375, -0.19, 0.035, 1., 0., 0., 0.], dims=[0.3, 0.01, 0.07]),  #x,y,z, qx, qy, qz, qw
    # Cuboid(name="bin", pose=[0.45, 0.0, 0.035, 1., 0., 0., 0.], dims=[0.3, 0.4, 0.07]),  #x,y,z, qx, qy, qz, qw
    Cuboid(name="table", pose=[0.0, 0.0, -0.05, 1., 0., 0., 0.], dims=[5.0, 5.0, 0.1]),  #x,y,z, qx, qy, qz, qw
    Cuboid(name="column1", pose=[0.84, 0.36, 0.5, 1., 0., 0., 0.], dims=[0.08, 0.08, 1.0]),
    Cuboid(name="column2", pose=[0.84, -0.36, 0.5, 1., 0., 0., 0.], dims=[0.08, 0.08, 1.0]),
    Cuboid(name="column3", pose=[-0.26, 0.36, 0.5, 1., 0., 0., 0.], dims=[0.08, 0.08, 1.0]),
    Cuboid(name="column4", pose=[-0.26, -0.36, 0.5, 1., 0., 0., 0.], dims=[0.08, 0.08, 1.0])]  #x,y,z, qx, qy, qz, qw

world_config = WorldConfig(cuboid=cuboids)

# create motion gen with a cuboid cache to be able to load obstacles later:
motion_gen_cfg = MotionGenConfig.load_from_robot_config(
    robot_config,
    world_config,
    self.tensor_args,
    trajopt_tsteps=34, # 34
    collision_cache = {"obb": 9},
    interpolation_steps=5000,
    num_ik_seeds=50,
    num_trajopt_seeds=6,
    grad_trajopt_iters=500,
    trajopt_dt=0.5,
    interpolation_dt=interpolation_dt,
    evaluate_interpolated_trajectory=True,
    js_trajopt_dt=0.5,
    js_trajopt_tsteps=34,  # 34
    use_cuda_graph=False,
    collision_activation_distance=collision_activation_distance)

self.motion_gen = MotionGen(motion_gen_cfg)
self.motion_gen.clear_world_cache()
self.motion_gen.warmup()
self.motion_gen.update_world(world_config)

if self.constrain_grasp_approach:
    pose_metric = PoseCostMetric.create_grasp_approach_metric(
        offset_position=0.05, tstep_fraction=0.2, linear_axis=2
    )
    self.plan_config = MotionGenPlanConfig(
        enable_graph=False,
        max_attempts=10,
        enable_graph_attempt=None,
        enable_finetune_trajopt=True,
        partial_ik_opt=False,
        parallel_finetune=False,
        pose_cost_metric=pose_metric,
    )

result = self.motion_gen.plan_single(q_start, goal_pose, self.plan_config.clone())
```

balakumar-s, 2024-04-26:
Is it possible for you to share the full python file including the imports?

maximiliangilles, 2024-04-29:
Hi @balakumar-s ,

sure. See attached my code:
```

# Third Party
import torch
import time
import copy
from flask import Flask, request, jsonify

# CuRobo
from curobo.geom.types import Cuboid, WorldConfig
from curobo.types.base import TensorDeviceType
from curobo.types.math import Pose
from curobo.types.robot import JointState
from curobo.util.usd_helper import UsdHelper, load_yaml
from curobo.wrap.reacher.motion_gen import MotionGen, MotionGenConfig, PoseCostMetric, MotionGenPlanConfig
from curobo.geom.sdf.world import CollisionCheckerType
from curobo.util_file import join_path, get_world_configs_path

app = Flask(__name__)

# In-memory storage for simplicity; in a real application, consider using a database.
data_store = {}

class MotionGenCUROBO:
    def __init__(self, constrain_grasp_approach: bool = False):
        self.constrain_grasp_approach = constrain_grasp_approach

        self.tensor_args = TensorDeviceType(device=torch.device("cuda:0"))

        interpolation_dt = 0.02
        collision_activation_distance = 0.02  # meters

        # create
        self.robot_config = load_yaml("/home/curobo/src/curobo/content/configs/robot/franka.yml")
        
        self.cuboids = [
            Cuboid(name="bin1", pose=[0.58, 0.0, 0.035, 1., 0., 0., 0.], dims=[0.01, 0.4, 0.07]),  #x,y,z, qx, qy, qz, qw
            Cuboid(name="bin2", pose=[0.295, 0.0, 0.035, 1., 0., 0., 0.], dims=[0.01, 0.4, 0.07]),  #x,y,z, qx, qy, qz, qw
            Cuboid(name="bin3", pose=[0.4375, 0.19, 0.035, 1., 0., 0., 0.], dims=[0.3, 0.01, 0.07]),  #x,y,z, qx, qy, qz, qw
            Cuboid(name="bin4", pose=[0.4375, -0.19, 0.035, 1., 0., 0., 0.], dims=[0.3, 0.01, 0.07]),  #x,y,z, qx, qy, qz, qw
            # Cuboid(name="bin", pose=[0.45, 0.0, 0.035, 1., 0., 0., 0.], dims=[0.3, 0.4, 0.07]),  #x,y,z, qx, qy, qz, qw
            Cuboid(name="table", pose=[0.0, 0.0, -0.05, 1., 0., 0., 0.], dims=[5.0, 5.0, 0.1]),  #x,y,z, qx, qy, qz, qw
            Cuboid(name="column1", pose=[0.84, 0.36, 0.5, 1., 0., 0., 0.], dims=[0.08, 0.08, 1.0]),
            Cuboid(name="column2", pose=[0.84, -0.36, 0.5, 1., 0., 0., 0.], dims=[0.08, 0.08, 1.0]),
            Cuboid(name="column3", pose=[-0.26, 0.36, 0.5, 1., 0., 0., 0.], dims=[0.08, 0.08, 1.0]),
            Cuboid(name="column4", pose=[-0.26, -0.36, 0.5, 1., 0., 0., 0.], dims=[0.08, 0.08, 1.0]),  #x,y,z, qx, qy, qz, qw
            Cuboid(name="slide", pose=[0.43, 0.36, 0.07, 1., 0., 0., 0.], dims=[0.18, 0.13, 0.14])]  #x,y,z, qx, qy, qz, qw

        world_config = WorldConfig(cuboid=self.cuboids)

        # create motion gen with a cuboid cache to be able to load obstacles later:
        motion_gen_cfg = MotionGenConfig.load_from_robot_config(
            self.robot_config,
            world_config,
            self.tensor_args,
            trajopt_tsteps=34, # 34
            collision_cache = {"obb": 9},
            interpolation_steps=5000,
            num_ik_seeds=50,
            num_trajopt_seeds=6,
            grad_trajopt_iters=500,
            trajopt_dt=0.5,
            interpolation_dt=interpolation_dt,
            evaluate_interpolated_trajectory=True,
            js_trajopt_dt=0.5,
            js_trajopt_tsteps=34,  # 34
            use_cuda_graph=True,
[... 191 lines cut ...]
    result = self.motion_gen.plan_single(q_start, goal_pose, self.plan_config.clone())
  File "/usr/local/lib/python3.10/dist-packages/curobo/wrap/reacher/motion_gen.py", line 1322, in plan_single
    result = self._plan_attempts(
  File "/usr/local/lib/python3.10/dist-packages/curobo/wrap/reacher/motion_gen.py", line 1097, in _plan_attempts
    valid_query = self.update_pose_cost_metric(
  File "/usr/local/lib/python3.10/dist-packages/curobo/wrap/reacher/motion_gen.py", line 2535, in update_pose_cost_metric
    [
  File "/usr/local/lib/python3.10/dist-packages/curobo/wrap/reacher/motion_gen.py", line 2536, in <listcomp>
    rollout.update_pose_cost_metric(metric)
  File "/usr/local/lib/python3.10/dist-packages/curobo/rollout/arm_reacher.py", line 467, in update_pose_cost_metric
    [
  File "/usr/local/lib/python3.10/dist-packages/curobo/rollout/arm_reacher.py", line 468, in <listcomp>
    x.update_offset_waypoint(
  File "/usr/local/lib/python3.10/dist-packages/curobo/rollout/cost/pose_cost.py", line 207, in update_offset_waypoint
    log_error(
  File "/usr/local/lib/python3.10/dist-packages/curobo/util/logger.py", line 45, in log_error
    raise
RuntimeError: No active exception to reraise
```

Again, thanks a lot for your support.

Best,

Maximilian

maximiliangilles, 2024-05-09:
Hi @balakumar-s,

I wanted to followup on this issue. Any idea or update, if this will be fixed in future releases?

Thanks a lot! I really appreciate your work and effort.

Best,

Max

balakumar-s, 2024-05-10:
Hi, does your code require any additional input to generate your error? 
I can debug by running your code but currently it looks like it needs additional information.

balakumar-s, 2024-07-22:
We have changed the graph so that warmup is not required before adding a waypoint. This should fix your issue. Please re-open if it doesn't.

closed 2024-07-22 by balakumar-s

maximiliangilles, 2024-07-23:
Thanks for letting me know, I will report when I run the script again.