NVlabs/curobo #228: RuntimeError: CUDA error: an illegal memory access was encountered Compile with `TORCH_USE_CUDA_DSA` to enable device-side assertions

lihsinn, 2024-04-19:
If it’s not a bug, please use discussions: https://github.com/NVlabs/curobo/discussions

Please provide the below information in addition to your issue:

1. cuRobo installation mode (choose from [python, isaac sim, docker python, docker isaac sim]): isaac sim
2. python version: Python 3.10.12
3. Isaac Sim version (if using):  isaac sim 2023.1.1

I ran the simple_stacking.py code and encountered a CUDA error when I changed it to my own robot.

------
`[0.040s] [ext: omni.kit.async_engine-0.0.0] startup
[0.220s] [ext: omni.activity.core-1.0.1] startup
[0.222s] [ext: omni.assets.plugins-0.0.0] startup
[0.223s] [ext: omni.stats-0.0.0] startup
[0.223s] [ext: omni.client-1.0.1] startup
[0.228s] [ext: omni.activity.profiler-1.0.2] startup
[0.229s] [ext: omni.gpu_foundation-0.0.0] startup
[0.234s] [ext: omni.rtx.shadercache.vulkan-1.0.0] startup
[0.235s] [ext: carb.windowing.plugins-1.0.0] startup
[0.239s] [ext: omni.kit.renderer.init-0.0.0] startup

|---------------------------------------------------------------------------------------------|
| Driver Version: 525.89.02     | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 4090          | Yes: 0 |     | 24564   MB | 10de      | 0          |
|     |                                  |        |     |            | 2684      | f723ef00.. |
|     |                                  |        |     |            | 1         |            |
|=============================================================================================|
| OS: 22.04.3 LTS (Jammy Jellyfish) ubuntu, Version: 22.04.3, Kernel: 5.15.0-1051-intel-iotg
| XServer Vendor: The X.Org Foundation, XServer Version: 12101004 (1.21.1.4)
| Processor: 13th Gen Intel(R) Core(TM) i7-13700KF | Cores: 16 | Logical: 32
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 128653 | Free Memory: 90257
| Total Page/Swap (MB): 2047 | Free Page/Swap: 2018
|---------------------------------------------------------------------------------------------|
[0.641s] [ext: omni.kit.pipapi-0.0.0] startup
[0.642s] [ext: omni.kit.pip_archive-0.0.0] startup
[0.642s] [ext: omni.pip.compute-1.2.0] startup
[0.642s] [ext: omni.pip.cloud-1.0.1] startup
[0.643s] [ext: omni.isaac.core_archive-2.2.1] startup
[0.643s] [ext: omni.pip.torch-2_0_1-2.0.2] startup
[0.664s] [ext: omni.kit.telemetry-0.5.0] startup
[0.680s] [ext: omni.kit.loop-isaac-1.1.0] startup
[0.681s] [ext: omni.kit.test-0.0.0] startup
[0.682s] [ext: omni.isaac.ml_archive-1.1.3] startup
[0.682s] [ext: omni.mtlx-0.1.0] startup
[0.682s] [ext: omni.usd.config-1.0.3] startup
[0.685s] [ext: omni.gpucompute.plugins-0.0.0] startup
[0.685s] [ext: omni.usd.libs-1.0.0] startup
[0.771s] [ext: omni.appwindow-1.1.5] startup
[1.135s] [ext: omni.kit.renderer.core-0.0.0] startup
[1.883s] [ext: omni.kit.renderer.capture-0.0.0] startup
[1.885s] [ext: omni.kit.renderer.imgui-0.0.0] startup
[1.931s] [ext: carb.audio-0.1.0] startup
[1.952s] [ext: omni.ui-2.18.5] startup
[1.960s] [ext: omni.uiaudio-1.0.0] startup
[1.961s] [ext: omni.kit.mainwindow-1.0.1] startup
[1.961s] [ext: omni.kit.uiapp-0.0.0] startup
[1.961s] [ext: omni.usd.schema.physics-0.0.0] startup
[2.054s] [ext: omni.usd.schema.audio-0.0.0] startup
[2.058s] [ext: omni.usd.schema.forcefield-0.0.0] startup
[2.062s] [ext: omni.usd.schema.omnigraph-1.0.0] startup
[2.066s] [ext: omni.usd.schema.scene.visualization-2.0.2] startup
[2.066s] [ext: omni.usd.schema.semantics-0.0.0] startup
[2.072s] [ext: omni.usd.schema.physx-0.0.0] startup
[... 341 lines cut ...]
    result = self._plan_from_solve_state(
  File "/home/lily/Desktop/20230605_Job_Techman_Project/01_Omniverse_IsaacSim/02_Robotics_Simulator/2024Q2_curobo/curobo/src/curobo/wrap/reacher/motion_gen.py", line 1511, in _plan_from_solve_state
    ik_result = self._solve_ik_from_solve_state(
  File "/home/lily/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/python/lib/python3.10/contextlib.py", line 79, in inner
    return func(*args, **kwds)
  File "/home/lily/Desktop/20230605_Job_Techman_Project/01_Omniverse_IsaacSim/02_Robotics_Simulator/2024Q2_curobo/curobo/src/curobo/wrap/reacher/motion_gen.py", line 1020, in _solve_ik_from_solve_state
    ik_result = self.ik_solver.solve_any(
  File "/home/lily/Desktop/20230605_Job_Techman_Project/01_Omniverse_IsaacSim/02_Robotics_Simulator/2024Q2_curobo/curobo/src/curobo/wrap/reacher/ik_solver.py", line 1160, in solve_any
    return self.solve_single(
  File "/home/lily/Desktop/20230605_Job_Techman_Project/01_Omniverse_IsaacSim/02_Robotics_Simulator/2024Q2_curobo/curobo/src/curobo/wrap/reacher/ik_solver.py", line 654, in solve_single
    return self._solve_from_solve_state(
  File "/home/lily/Desktop/20230605_Job_Techman_Project/01_Omniverse_IsaacSim/02_Robotics_Simulator/2024Q2_curobo/curobo/src/curobo/wrap/reacher/ik_solver.py", line 1043, in _solve_from_solve_state
    result = self.solver.solve(goal_buffer, coord_position_seed)
  File "/home/lily/Desktop/20230605_Job_Techman_Project/01_Omniverse_IsaacSim/02_Robotics_Simulator/2024Q2_curobo/curobo/src/curobo/wrap/wrap_base.py", line 147, in solve
    act_seq = self.optimize(seed, shift_steps=0)
  File "/home/lily/Desktop/20230605_Job_Techman_Project/01_Omniverse_IsaacSim/02_Robotics_Simulator/2024Q2_curobo/curobo/src/curobo/wrap/wrap_base.py", line 71, in optimize
    act_seq = opt.optimize(act_seq, shift_steps)
  File "/home/lily/Desktop/20230605_Job_Techman_Project/01_Omniverse_IsaacSim/02_Robotics_Simulator/2024Q2_curobo/curobo/src/curobo/opt/opt_base.py", line 170, in optimize
    out = self._optimize(opt_tensor, shift_steps, n_iters)
  File "/home/lily/Desktop/20230605_Job_Techman_Project/01_Omniverse_IsaacSim/02_Robotics_Simulator/2024Q2_curobo/curobo/src/curobo/opt/newton/newton_base.py", line 146, in _optimize
    best_q, best_cost, q, grad_q = self._call_opt_iters_graph(q, grad_q)
  File "/home/lily/Desktop/20230605_Job_Techman_Project/01_Omniverse_IsaacSim/02_Robotics_Simulator/2024Q2_curobo/curobo/src/curobo/opt/newton/newton_base.py", line 521, in _call_opt_iters_graph
    self._cu_opt_q.clone(),
RuntimeError: CUDA error: an illegal memory access was encountered
Compile with `TORCH_USE_CUDA_DSA` to enable device-side assertions.

balakumar-s, 2024-04-26:
Did you make any changes to simple_stacking.py?

lihsinn, 2024-04-29:
Change the 'Franka' class to my robot class and update the related YAML file. Also, comment out the lines 'articulation_controller.set_gains' and 'articulation_controller.set_max_efforts', that's all!

balakumar-s, 2024-05-15:
Does your robot work with other examples like `examples/isaac_sim/motion_gen_reacher.py` ?

frederiknolte, 2024-07-22:
I am having the same problem. I also replaced the Franka Panda with a different robot but even if I put the Panda back in it still does not work. Interestingly, the bug is not consistent. It sometimes happens, and sometimes doesn't.

balakumar-s, 2024-07-22:
Have you tried with the latest commit? We pushed a CUDA kernel fix that might resolve this issue.

frederiknolte, 2024-07-23:
The problem came back. Could you please reopen the issue?
I first ran my code without problems. Then I wanted to check the result of an earlier computation in the code and restarted the execution and the bug appeared again. Nothing changed in-between the two executions.

balakumar-s, 2024-07-23:
After pulling the latest main, did you run `pip install -e . --no-build-isolation`?

frederiknolte, 2024-07-23:
My dockerfile is based on the one provided in this repo, so the command is `pip install .[dev] --no-build-isolation`.

Also, I played around with the IK solver and noticed that when specifying the problem shape incorrectly (e.g. having a target position of shape `[5, 100, 3]`) and calling a solve function that only supports a different shape (in this case `solve_batch_env`), there is no meaningful check or error message but it goes straight to illegal memory access.

I restarted the docker container and it works again (if the shapes are correct of course).