NVlabs/curobo #230: timeout in MotionGenPlanConfig not strictly enforced

Cfather, 2024-04-19:
Please provide the below information in addition to your issue:

1. cuRobo installation mode: python
2. python version: 3.10
3. 

------
Issue Details

I used `examples/motion_gen_example.py` to test timeout.
I change the world yaml (`collision_table.yml`) that the obstacle is so large that there won't be a feasible solution for whatever start and goal poses.
I set `timeout` in `MotionGenPlanConfig` to be 10.0 seconds by default.
I set `max_attempts` in `MotionGenPlanConfig` to be a very large number (1000 or 10000) so that it will keep trying since the problem is always infeasible. 
Ideally, **it should trigger the timeout condition and exit early before it reaches `max_attempts`**. The solve time should approximately be equal to 10.0 seconds. But instead, curobo takes a longer time to return the infeasible result. The output is attached below:
```
Trajectory Generated:  tensor([False], device='cuda:0') 21.21181082725525 TrajOpt Fail tensor([0.1000, 0.1000], device='cuda:0')
```
`21.21181082725525` is the actual solve time.
Other parameters in `MotionGenPlanConfig`: `parallel_finetune=True, enable_finetune_trajopt=True`

balakumar-s, 2024-04-22:
Can you share the changes you made to `collision_table.yml` so that we can reproduce this?

Cfather, 2024-04-23:
Thanks for the reply!

I simply changed the size (dim) to 500, so that the table covers the entire reachable region of the robot.