NVlabs/curobo #235: Installation problems for building with nvblox

wuyingnan, 2024-04-25:
I am installing curobo with nvblox following the documents. (https://curobo.org/get_started/1_install_instructions.html#installing-nvblox-for-precxx11-abi-and-isaac-sim)

I got two problems following the installing procedure.

1. Step 7 install nvblox. The command is `cd ${PKGS_PATH} &&  git clone https://github.com/valtsblukis/nvblox.git && cd ${PKGS_PATH}/nvblox/nvblox mkdir build && cd build && \
cmake ..  -DBUILD_REDISTRIBUTABLE=ON \
-DCMAKE_CXX_FLAGS=-D_GLIBCXX_USE_CXX11_ABI=0  -DPRE_CXX11_ABI_LINKABLE=ON \
-DSQLITE3_BASE_PATH="${PKGS_PATH}/sqlite/install/" -DGLOG_BASE_PATH="${PKGS_PATH}/glog/install/" \
-DGFLAGS_BASE_PATH="${PKGS_PATH}/gflags/install/" -DCMAKE_CUDA_FLAGS=-D_GLIBCXX_USE_CXX11_ABI=0 && \
make -j32 && \
sudo make install
` There is a "&&" missing between "cd ${PKGS_PATH}/nvblox/nvblox" and "mkdir build"
2. Glog is compiled as static, while it links against a system library "libunwind.so". In nvblox linking procedure, the linker will complain that it cannot find some functions. I had to manually add libunwind.so into GLOG_LIBRARIES in nvblox CMakeLists.txt.

balakumar-s, 2024-04-26:
Thanks for letting us know! 
After doing these fixes, are you able to use nvblox?

wuyingnan, 2024-05-07:
Yes. I have run the nvblox collision checker demo.

closed 2024-05-15 by balakumar-s