NVlabs/curobo #262: Getting Started Using in Python Examples Not Working

tylerlum, 2024-05-10:
Small documentation issues for the getting started that could be improved here, since it doesn't "just work" as it is currently written: https://curobo.org/get_started/2a_python_examples.html

### Forward Kinematics
Given example code:

```
# Third Party
import torch

# cuRobo
from curobo.cuda_robot_model.cuda_robot_model import CudaRobotModel, CudaRobotModelConfig
from curobo.types.base import TensorDeviceType
from curobo.types.robot import RobotConfig
from curobo.util_file import get_robot_path, join_path, load_yaml

# convenience function to store tensor type and device
tensor_args = TensorDeviceType()

# this example loads urdf from a configuration file, you can also load from path directly
# load a urdf, the base frame and the end-effector frame:
config_file = load_yaml(join_path(get_robot_path(), "franka.yml"))

urdf_file = config_file["robot_cfg"]["kinematics"][
    "urdf_path"
]  # Send global path starting with "/"
base_link = config_file["robot_cfg"]["kinematics"]["base_link"]
ee_link = config_file["robot_cfg"]["kinematics"]["ee_link"]

# Generate robot configuration from  urdf path, base frame, end effector frame

robot_cfg = RobotConfig.from_basic(urdf_file, base_link, ee_link, tensor_args)

kin_model = CudaRobotModel(robot_cfg.kinematics)

# compute forward kinematics:
# torch random sampling might give values out of joint limits
q = torch.rand((10, kin_model.get_dof()), **vars(tensor_args))
out = kin_model.get_state(q)
```

Error:
```
  File "test_fk.py", line 20, in <module>
    q = torch.rand((10, kin_model.get_dof()), **vars(tensor_args))
TypeError: rand() received an invalid combination of arguments - got (tuple, collision_distance_dtype=torch.dtype, collision_gradient_dtype=torch.dtype, collision_geometry_dtype=torch.dtype, dtype=torch.dtype, device=torch.device), but expected one of:
 * (tuple of ints size, *, torch.Generator generator, tuple of names names, torch.dtype dtype, torch.layout layout, torch.device device, bool pin_memory, bool requires_grad)
 * (tuple of ints size, *, torch.Generator generator, Tensor out, torch.dtype dtype, torch.layout layout, torch.device device, bool pin_memory, bool requires_grad)
 * (tuple of ints size, *, Tensor out, torch.dtype dtype, torch.layout layout, torch.device device, bool pin_memory, bool requires_grad)
 * (tuple of ints size, *, tuple of names names, torch.dtype dtype, torch.layout layout, torch.device device, bool pin_memory, bool requires_grad)
```

Required change:

Replace this:
```
q = torch.rand((10, kin_model.get_dof()), **vars(tensor_args))
```

With this:
```
q = torch.rand((10, kin_model.get_dof()), **(tensor_args.as_torch_dict()))
```

Should make this change to all examples with var(tensor_args) used in this way (there's another one in forward kinematics)

### Collision Checking

Example:
```
# Third Party
import torch
# cuRobo
from curobo.types.base import TensorDeviceType
from curobo.wrap.model.robot_world import RobotWorld, RobotWorldConfig

robot_file = "franka.yml"

# create a world from a dictionary of objects
# cuboid: {} # dictionary of objects that are cuboids
# mesh: {} # dictionary of objects that are meshes
world_config = {
    "cuboid": {
        "table": {"dims": [2, 2, 0.2], "pose": [0.4, 0.0, -0.1, 1, 0, 0, 0]},
        "cube_1": {"dims": [0.1, 0.1, 0.2], "pose": [0.4, 0.0, 0.5, 1, 0, 0, 0]},
    },
    "mesh": {
        "scene": {
            "pose": [1.5, 0.080, 1.6, 0.043, -0.471, 0.284, 0.834],
            "file_path": "scene/nvblox/srl_ur10_bins.obj",
        }
    },
}
tensor_args = TensorDeviceType()
config = RobotWorldConfig.load_from_config(robot_file, world_file,
                                          collision_activation_distance=0.0)
curobo_fn = RobotWorld(config)
```

Error:
```
    config = RobotWorldConfig.load_from_config(robot_file, world_file,
NameError: name 'world_file' is not defined
```

Simply need to change `world_file` to `world_cfg`

Please provide the below information in addition to your issue:

1. cuRobo installation mode (choose from [python, isaac sim, docker python, docker isaac sim]): python
2. python version: Python 3.8.19
3. Isaac Sim version (if using): Not used

------
Issue Details

balakumar-s, 2024-05-17:
Thanks! we will fix this

balakumar-s, 2024-07-22:
Fixed

closed 2024-07-22 by balakumar-s