NVlabs/curobo #268: Occasional bug with `g_dim` shape

tylerlum, 2024-05-10:
In `src/curobo/wrap/reacher/motion_gen.py`:
```
                    g_dim = torch.nonzero(ik_success).view(-1)[graph_result.success]

                    self._batch_graph_search_buffer.copy_at_index(
                        graph_result.interpolated_plan, g_dim
                    )

                    # result.graph_plan = JointState.from_position(
                    #    self._batch_graph_search_buffer,
                    #    joint_names=graph_result.interpolated_plan.joint_names,
                    # )
                    result.interpolated_plan = self._batch_graph_search_buffer
                    g_dim = g_dim.cpu().squeeze().tolist()
                    for x, x_val in enumerate(g_dim):
                        self._batch_path_buffer_last_tstep[x_val] = (
                            graph_result.path_buffer_last_tstep[x]
                        )
 ```

I have occasionally seen a bug where it says something like:
```
TypeError: 'int' (maybe float I forgot) object is not iterable
```

Which comes from:
```
                    g_dim = g_dim.cpu().squeeze().tolist()
                    for x, x_val in enumerate(g_dim):
```

I guess there is some weird case where g_dim is an int instead of a list. Sorry I can't reproduce it right now, but my hack fix was:

```
                    g_dim = g_dim.cpu().squeeze().tolist()
+                    # Add to handle occasional case when g_dim isn't a list
+                    if isinstance(g_dim, int):
+                        g_dim = [g_dim]
                    for x, x_val in enumerate(g_dim):
```

The ideal solution is to understand the desired and actual shapes of:
```
                    g_dim = torch.nonzero(ik_success).view(-1)[graph_result.success]
                    g_dim = g_dim.cpu().squeeze().tolist()
```

Then fix it. My guess is that squeeze().tolist() on a tensor of shape (1,) or (1, 1) gives a value instead of a list, but maybe Balakumar knows the shapes and can update the squeeze appropriately. Could also be somethign related to `torch.nonzero(ik_success).view(-1)[graph_result.success]`

1. cuRobo installation mode (choose from [python, isaac sim, docker python, docker isaac sim]): python
2. python version: 3.8.19
3. Isaac Sim version (if using): Not used

------
Issue Details

balakumar-s, 2024-05-12:
What pyTorch version are you using?

tylerlum, 2024-05-12:
I'm using:
```
>>> import torch
>>> torch.__version__
'2.1.2+cu118'
```

balakumar-s, 2024-05-15:
We will add your suggestion.

closed 2024-05-15 by balakumar-s