NVlabs/curobo #284: Error installing CUDA

Sajjan2406, 2024-05-27:
1. cuRobo installation mode (choose from [python, isaac sim, docker python, docker isaac sim]): IssacSim
2. python version:
3. Isaac Sim version (if using): 2023.1.0-hotfix.1

------
Issue Details
i am not able to install CUDA 11.7 or 11.8. Right now i am using CUDA version 12.2, NVIDIA driver 535 and Ubuntu 22 04. When i try to install 11.7 or 11.8, it removes the present nvidia smi and gives out the following error: E: Sub-process /usr/bin/dpkg returned an error code (1) 
i also tried installing the drivers compitable with 11.7 and 11.8. But facing the same issue. 
is it also possible to install curobo library in the above mentioned spec of cuda and the nvidia drivers?
Please suggest me how to proceed with this!

Sepirence, 2024-05-28:
I think that is not related with curobo library. You can solve that dpkg error with searching your problem on internet.

By the way, I am using curobo very well with nvidia_driver 535.171.04, CUDA 11.8, ubuntu 22.04 and Isaac sim 2023.1.0. So you do not have to worry about compitability issue of this library. I think you can solve your package issue enough.

Sajjan2406, 2024-06-06:
@Sepirence thanks for the reply. I tried many ways to solve it, but it did not work out in my system. Right now i am using the Docker Container to use cuRobo. 

Thanks again!

closed 2024-06-06 by Sajjan2406