NVlabs/curobo #287: Installing curobo

FrankSun1231, 2024-05-30:
I already successfully installed the CUDA11.5.
when I try to do this command: pip install -e . --no-build-isolation

Building wheels for collected packages: nvidia_curobo
  Building editable for nvidia_curobo (pyproject.toml) ... error
  error: subprocess-exited-with-error

  × Building editable for nvidia_curobo (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [147 lines of output]

              ********************************************************************************
              An error happened while installing `nvidia_curobo` in editable mode.

              The following steps are recommended to help debug this problem:

              - Try to install the project normally, without using the editable mode.
                Does the error still persist?
                (If it does, try fixing the problem before attempting the editable mode).
              - If you are using binary extensions, make sure you have all OS-level
                dependencies installed (e.g. compilers, toolchains, binary libraries, ...).
              - Try the latest version of setuptools (maybe the error was already fixed).
              - If you (or your project dependencies) are using any setuptools extension
                or customization, make sure they support the editable mode.

              After following the steps above, if the problem still persists and
              you think this is related to how setuptools handles editable installations,
              please submit a reproducible example
              (see https://stackoverflow.com/help/minimal-reproducible-example) to:

                  https://github.com/pypa/setuptools/issues

              See https://setuptools.pypa.io/en/latest/userguide/development_mode.html for details.
              ********************************************************************************

      !!
        cmd_obj.run()
      error: command '/usr/bin/gcc' failed with exit code 1
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building editable for nvidia-curobo
Failed to build nvidia-curobo
ERROR: Could not build wheels for nvidia-curobo, which is required to install pyproject.toml-based projects

device: lenovo R9000p2021 wsl
Nvidia GeForce RTX 3060

nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2021 NVIDIA Corporation
Built on Thu_Nov_18_09:45:30_PST_2021
Cuda compilation tools, release 11.5, V11.5.119
Build cuda_11.5.r11.5/compiler.30672275_0

python -c "import torch; print(torch.__version__)"
2.3.0+rocm6.0

blooop, 2024-05-30:
```
RuntimeError:
  The detected CUDA version (11.5) mismatches the version that was used to compile
  PyTorch (12.1). Please make sure to use the same CUDA versions.
```

It looks like you need update your CUDA version.

Also see:
https://github.com/NVlabs/curobo/issues/282#issuecomment-2129713296

FrankSun1231, 2024-05-30:
Thanks for your reply. I just installed the CUDA11.8 and the torch 2.21. but there a new problem happened with same command: pip install -e . --no-build-isolation seems this time is not the cuda and pytorch problem. 

#note for cuda, pytorch, setup tools and wheels
python -c "import torch; print(torch.__version__)"
2.2.1+cu118
nvcc --version
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2022 NVIDIA Corporation
Built on Wed_Sep_21_10:33:58_PDT_2022
Cuda compilation tools, release 11.8, V11.8.89
Build cuda_11.8.r11.8/compiler.31833905_0

pip show setuptools:
Name: setuptools
Version: 70.0.0
Summary: Easily download, build, install, upgrade, and uninstall Python packages
Home-page: https://github.com/pypa/setuptools
Author: Python Packaging Authority
Author-email: distutils-sig@python.org
License: 
Location: /home/dengws/anaconda3/envs/cse571project/lib/python3.11/site-packages
Requires: 
Required-by: action-tutorials-py, ament-copyright, ament-cppcheck, ament-cpplint, ament-flake8, ament-index-python, ament-lint, ament-lint-cmake, ament-pep257, ament-uncrustify, ament-xmllint, demo-nodes-py, domain-coordinator, examples-rclpy-executors, examples-rclpy-minimal-action-client, examples-rclpy-minimal-action-server, examples-rclpy-minimal-client, examples-rclpy-minimal-publisher, examples-rclpy-minimal-service, examples-rclpy-minimal-subscriber, joint-state-publisher, joint-state-publisher-gui, launch, launch-ros, launch-testing, launch-testing-ros, launch-xml, launch-yaml, quality-of-service-demo-py, rqt-action, rqt-bag, rqt-bag-plugins, rqt-console, rqt-controller-manager, rqt-gauges, rqt-graph, rqt-gui, rqt-gui-py, rqt-joint-trajectory-controller, rqt-moveit, rqt-msg, rqt-plot, rqt-publisher, rqt-py-console, rqt-reconfigure, rqt-robot-dashboard, rqt-robot-monitor, rqt-robot-steering, rqt-runtime-monitor, rqt-service-caller, rqt-shell, rqt-srv, rqt-tf-tree, rqt-topic, sensor-msgs-py, sros2, teleop-twist-keyboard, tf2-ros-py, tf2-tools, topic-monitor

pip show wheel:
Name: wheel
Version: 0.43.0
Summary: A built-package format for Python
Home-page: 
Author: 
Author-email: Daniel Holth <dholth@fastmail.fm>
License: 
Location: /home/dengws/anaconda3/envs/cse571project/lib/python3.11/site-packages
Requires: 
Required-by: 

#note for cuda, pytorch, setup tools and wheels

# typical problem:         
File "/home/dengws/anaconda3/envs/cse571project/lib/python3.11/site-packages/torch/utils/cpp_extension.py", line 1976, in _get_cuda_arch_flags
          arch_list[-1] += '+PTX'
          ~~~~~~~~~^^^^
      IndexError: list index out of range
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building editable for nvidia_curobo
Failed to build nvidia_curobo
ERROR: Could not build wheels for nvidia_curobo, which is required to install pyproject.toml-based projects

# whole problem view below.

(cse571project) dengws@dengws--R9000P2021H:~/Desktop/571project/curobo$ pip install -e . --no-build-isolation
Obtaining file:///home/dengws/Desktop/571project/curobo
  Checking if build backend supports build_editable ... done
  Preparing editable metadata (pyproject.toml) ... done
Collecting pybind11 (from nvidia_curobo==0.0.0)
  Using cached pybind11-2.12.0-py3-none-any.whl.metadata (9.5 kB)
Requirement already satisfied: networkx in /home/dengws/anaconda3/envs/cse571project/lib/python3.11/site-packages (from nvidia_curobo==0.0.0) (3.3)
Requirement already satisfied: numpy in /home/dengws/anaconda3/envs/cse571project/lib/python3.11/site-packages (from nvidia_curobo==0.0.0) (1.26.4)
Collecting numpy-quaternion (from nvidia_curobo==0.0.0)
  Using cached numpy_quaternion-2023.0.3-cp311-cp311-manylinux_2_5_x86_64.manylinux1_x86_64.manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (14 kB)
Collecting pyyaml (from nvidia_curobo==0.0.0)
  Using cached PyYAML-6.0.1-cp311-cp311-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (2.1 kB)
Collecting setuptools-scm>=6.2 (from nvidia_curobo==0.0.0)
  Using cached setuptools_scm-8.1.0-py3-none-any.whl.metadata (6.6 kB)
Requirement already satisfied: torch>=1.10 in /home/dengws/anaconda3/envs/cse571project/lib/python3.11/site-packages (from nvidia_curobo==0.0.0) (2.2.1+cu118)
Collecting trimesh (from nvidia_curobo==0.0.0)
  Using cached trimesh-4.4.0-py3-none-any.whl.metadata (18 kB)
Collecting yourdfpy>=0.0.53 (from nvidia_curobo==0.0.0)
[... 312 lines cut ...]
        File "/home/dengws/anaconda3/envs/cse571project/lib/python3.11/site-packages/setuptools/_distutils/command/build_ext.py", line 505, in _build_extensions_serial
          self.build_extension(ext)
        File "/home/dengws/anaconda3/envs/cse571project/lib/python3.11/site-packages/setuptools/command/build_ext.py", line 252, in build_extension
          _build_ext.build_extension(self, ext)
        File "/home/dengws/anaconda3/envs/cse571project/lib/python3.11/site-packages/setuptools/_distutils/command/build_ext.py", line 560, in build_extension
          objects = self.compiler.compile(
                    ^^^^^^^^^^^^^^^^^^^^^^
        File "/home/dengws/anaconda3/envs/cse571project/lib/python3.11/site-packages/setuptools/_distutils/ccompiler.py", line 600, in compile
          self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts)
        File "/home/dengws/anaconda3/envs/cse571project/lib/python3.11/site-packages/torch/utils/cpp_extension.py", line 607, in unix_wrap_single_compile
[... cut ...]
        File "/home/dengws/anaconda3/envs/cse571project/lib/python3.11/site-packages/torch/utils/cpp_extension.py", line 1976, in _get_cuda_arch_flags
          arch_list[-1] += '+PTX'
          ~~~~~~~~~^^^^
      IndexError: list index out of range
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building editable for nvidia_curobo
Failed to build nvidia_curobo
ERROR: Could not build wheels for nvidia_curobo, which is required to install pyproject.toml-based projects

balakumar-s, 2024-05-30:
- cuRobo is not supported in python 3.11. Can you use 3.10? 
- There is some error in finding NVML from pytorch:
```
  gcc -pthread -B /home/dengws/anaconda3/envs/cse571project/compiler_compat -DNDEBUG -fwrapv -O2 -Wall -fPIC -O2 -isystem /home/dengws/anaconda3/envs/cse571project/include -fPIC -O2 -isystem /home/dengws/anaconda3/envs/cse571project/include -fPIC -I/home/dengws/anaconda3/envs/cse571project/lib/python3.11/site-packages/torch/include -I/home/dengws/anaconda3/envs/cse571project/lib/python3.11/site-packages/torch/include/torch/csrc/api/include -I/home/dengws/anaconda3/envs/cse571project/lib/python3.11/site-packages/torch/include/TH -I/home/dengws/anaconda3/envs/cse571project/lib/python3.11/site-packages/torch/include/THC -I/usr/local/cuda-11.8/include -I/home/dengws/anaconda3/envs/cse571project/include/python3.11 -c src/curobo/curobolib/cpp/lbfgs_step_cuda.cpp -o /tmp/tmp_qiqvowk.build-temp/src/curobo/curobolib/cpp/lbfgs_step_cuda.o -DTORCH_API_INCLUDE_EXTENSION_H -DPYBIND11_COMPILER_TYPE=\"_gcc\" -DPYBIND11_STDLIB=\"_libstdcpp\" -DPYBIND11_BUILD_ABI=\"_cxxabi1011\" -DTORCH_EXTENSION_NAME=lbfgs_step_cu -D_GLIBCXX_USE_CXX11_ABI=0 -std=c++17
  /home/dengws/anaconda3/envs/cse571project/lib/python3.11/site-packages/torch/cuda/__init__.py:628: UserWarning: Can't initialize NVML
    warnings.warn("Can't initialize NVML")
```

Can you try if the below function work?

```
import torch
a = torch.zeros(5).cuda()
```

closed 2024-06-24 by balakumar-s