NVlabs/curobo #289: ModuleNotFoundError: No module named 'cv2'

Tranthanhbao198, 2024-05-31:
If it’s not a bug, please use discussions: https://github.com/NVlabs/curobo/discussions

Please provide the below information in addition to your issue:

1. cuRobo installation mode (choose from [python, isaac sim, docker python, docker isaac sim]): IsaacSim
2. python version: in conda env python 3.9
3. Isaac Sim version (if using): 2022.2.1

------
hello, i got this error. I installed opencv-python both in conda env (PnP) and base, but i always got error with cv2 when i ran omni_python realsense_viewer.py. Please help me to fix it. thank you so much
[image]

Tranthanhbao198, 2024-05-31:
[image]

balakumar-s, 2024-05-31:
Can you try: `omni_python -m pip install opencv-python`?

Tranthanhbao198, 2024-05-31:
> Can you try: `omni_python -m pip install opencv-python`?

hello, i am reinstalling, but at step8, i got this error, can you help me?
[image]

Tranthanhbao198, 2024-05-31:
and this error

> Can you try: `omni_python -m pip install opencv-python`?

yes thank you it worked

Tranthanhbao198, 2024-05-31:
3.  [14%] Built target ext_eigen
[ 15%] Building CUDA object CMakeFiles/nvblox_gpu_hash.dir/src/core/error_check.cu.o
[ 16%] Building CXX object CMakeFiles/nvblox_gpu_hash.dir/src/utils/nvtx_ranges.cpp.o
[ 17%] Building CXX object CMakeFiles/nvblox_gpu_hash.dir/src/utils/timing.cpp.o
[ 20%] Building CUDA object CMakeFiles/nvblox_gpu_hash.dir/src/gpu_hash/gpu_layer_view.cu.o
[ 20%] Building CUDA object CMakeFiles/nvblox_gpu_hash.dir/src/gpu_hash/gpu_set.cu.o
nvcc fatal   : Unsupported gpu architecture 'compute_89'
make[2]: *** [CMakeFiles/nvblox_gpu_hash.dir/build.make:135: CMakeFiles/nvblox_gpu_hash.dir/src/gpu_hash/gpu_set.cu.o] Error 1
make[2]: *** Waiting for unfinished jobs....
nvcc fatal   : Unsupported gpu architecture 'compute_89'
make[2]: *** [CMakeFiles/nvblox_gpu_hash.dir/build.make:77: CMakeFiles/nvblox_gpu_hash.dir/src/core/error_check.cu.o] Error 1
nvcc fatal   : Unsupported gpu architecture 'compute_89'
make[2]: *** [CMakeFiles/nvblox_gpu_hash.dir/build.make:120: CMakeFiles/nvblox_gpu_hash.dir/src/gpu_hash/gpu_layer_view.cu.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:261: CMakeFiles/nvblox_gpu_hash.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

**--> this error when i run step 7 in Installing nvblox for PRECXX11_ABI and Isaac Sim**
[image]

Tranthanhbao198, 2024-05-31:
@balakumar-s bala
 please help me to fix them. thank you so much

balakumar-s, 2024-05-31:
Is it possible for you to use the provided docker?

balakumar-s, 2024-05-31:
In addition, python version in conda should match isaac sim python version of 3.8.

Also, check your cuda version. It should be 11.8.

Tranthanhbao198, 2024-06-01:
> Is it possible for you to use the provided docker?

Hello, i dont have many experiment with docker@.@

Tranthanhbao198, 2024-06-01:
> In addition, python version in conda should match isaac sim python version of 3.8.
> 
> Also, check your cuda version. It should be 11.8.
How do I know version of python isaacsim? Because I don't see any information about it before? Please let me know.
And, I read docs and it said that I should install cuda11.7, so I should need to reinstall it? 
Thank you so much.