NVlabs/curobo #290: mpc fail to plan with mpc_example.py

MarcoPereira00, 2024-05-31:
1. cuRobo installation mode: docker isaac sim
2. python version: 3.10.13
3. Isaac Sim version (if using): 2023.1.1

Issue Details
When I run the example mpc_example.py, the robot just can't go to the target and falls to the ground

https://github.com/NVlabs/curobo/assets/75946345/2c8154c9-4efa-440e-8cfc-afaef6d6e760

balakumar-s, 2024-05-31:
Try ur5e robot instead.

We don't have all robots working in isaac sim 2023.1.1. Use 2023.1.0.

MarcoPereira00, 2024-06-04:
Which version was used in the example where franka panda is used? isaac yes 2023.1.0?
Can you tell me which robots are working on isaac sim 2023.1.1 and how long will it take for franka panda to work?
How can I change the size of the “camera window” that represents obstacles?
Thanks

balakumar-s, 2024-06-24:
The examples use 2022.2.0. We have also tested with 2023.1.0. We do not have a timeline for supporting more robots through URDF in newer versions of Isaac sim. The key limitation is in how we import the robot from URDF. If you instead import the robot directly in Isaac sim and connect cuRobo to it, Franka Panda would work in 2023.1.1.

What do you mean by "camera window"?