NVlabs/curobo #291: Trajectory of motion generation

Entongsu, 2024-05-31:
Hi,

I've encountered trajectory generation for the Franka robot. During recent operations, I observed that the robot consistently overshoots the intended target position and then corrects its path by moving backward. I am not sure how to fix this problem. Could you give me some hints over that?

balakumar-s, 2024-06-01:
What commit of curobo are you using?

Entongsu, 2024-06-01:
I think I have used this version 1e0b5a80c295fb91bc4af97dc219504cba8a6f16. I have tried the newest one and got the error of ` error[~success] += 10000.0
RuntimeError: CUDA error: an illegal memory access was encountered
CUDA kernel errors might be asynchronously reported at some other API call, so the stacktrace below might be incorrect.
For debugging consider passing CUDA_LAUNCH_BLOCKING=1.
Compile with `TORCH_USE_CUDA_DSA` to enable device-side assertions.
`

Entongsu, 2024-06-01:
I have tried version 3bfed9d7737fc3865595e4a50b1fe84a32cf627f. The generated trajectories are worse than before.

balakumar-s, 2024-06-01:
Are you using batched planning or single pose planning?

Entongsu, 2024-06-01:
I use single-pose planning. Could you provide some recommended settings for the parameter? Thank you. ` motion_gen_config = MotionGenConfig.load_from_robot_config(
            robot_file,
            world_file,
            self.tensor_args,
            # trajopt_tsteps=32,
            # collision_checker_type=CollisionCheckerType.PRIMITIVE,
            # use_cuda_graph=True,
            # num_trajopt_seeds=4,
            # num_graph_seeds=4,
            # evaluate_interpolated_trajectory=True,
            interpolation_dt=0.01,
        )`

Entongsu, 2024-06-01:
I have found the problem; it may stem from the collision mesh I used for the motion generation. I have set something like this before. `  cuboids = [
                Cuboid(name="cube",
                       pose=np.concatenate([
                           translation_r2cube, quaternion_r2cube[[3, 0, 1, 2]]
                       ]),
                       dims=np.array(self.env.cube.size)*2)
            ]`. After I change the dims from `np.array(self.env.cube.size)*2` to `np.array(self.env.cube.size)`, it becomes much more better. For the `dims` of `cuboids`, is it the full size or half size of the cube?

balakumar-s, 2024-06-01:
Dims is full size. Also make sure finetune_trajopt is enabled in MotionGenPlanConfig.

balakumar-s, 2024-06-24:
Are you still having issues? If yes, please share a small script that we can reproduce.

closed 2024-06-25 by balakumar-s