NVlabs/curobo #294: Erroe whilw runnning python scripts

Sajjan2406, 2024-06-06:
1. cuRobo installation mode (choose from [python, isaac sim, docker python, docker isaac sim]):docker isaac sim
2. python version:3.8.10
3. Isaac Sim version (if using): 

------
Issue Details

When i run the example scripts, i get the following error :
[Error] [omni.appwindow.plugin] Failed to acquire IWindowing interface

PS: I am running headless in the docker container and the Streaming client on the Host machine

shengyangzhuang, 2024-07-10:
Same for me. Is there a solution?

Sajjan2406, 2024-07-12:
you need to provide the permission for the network connection on the local machine.
Try running: xhost +local:root on the local machine
The output would be 'non-network local connections being added to access control list '

closed 2025-04-25 by balakumar-s