NVlabs/curobo #295: Breaking Change with warp.Kernel

michaelGentner, 2024-06-07:
Hi,

I think that commit 0c51dd2 introduced a breaking change for usage with isaac sim 2022.2.1, 2023.1.0, 2023.1.1. since isaac is shipped with an older version of warp, which has a different interface for warp.Kernel.__init__() than the newest version.

https://github.com/NVlabs/curobo/blob/0c51dd2da8d4a76ebaa0dee4e302ef88e504bce0/src/curobo/rollout/cost/dist_cost.py#L227

Error (FYI usually it also expects the key field but that we added in the code when debugging):
[image]

Could also be that this is due to my container setup, but using the previous commit fixed the issue for me.

Additional info:
1. cuRobo installation mode: docker isaac-sim
2. python version: 
3. Isaac Sim version (if using): 2022.2.1, 2023.1.0, 2023.1.1

------

AndyZe, 2024-06-16:
I have the same issue on a bare metal installation, so it's not your docker setup  :+1: Thanks for the hint.

balakumar-s, 2024-06-18:
We recently updated to a newer version of warp that allows for more runtime kernel customization. This unfortunately broke isaac sim examples. We are working on getting our examples to work in isaac sim 4.0 which has the newest warp. Meanwhile, use an older version of curobo (v0.7.2 or lower).

balakumar-s, 2024-06-24:
Fixed in latest main branch.

closed 2024-06-24 by balakumar-s