NVlabs/curobo #301: Couldn't find function overload for `mesh_query_point`

AndyZe, 2024-06-16:
1. cuRobo installation mode (choose from [python, isaac sim, docker python, docker isaac sim]): isaac sim
2. python version: 3.10.12
3. Isaac Sim version (if using): 2023.1.1

Occurs when running `omni_python examples/isaac_sim/motion_gen_reacher.py`, after fixing #295.

------
```
Traceback (most recent call last):
  File "/home/andy/ws_nvidia/src/curobo/examples/isaac_sim/motion_gen_reacher.py", line 437, in <module>
    main()
  File "/home/andy/ws_nvidia/src/curobo/examples/isaac_sim/motion_gen_reacher.py", line 229, in main
    motion_gen.warmup(enable_graph=True, warmup_js_trajopt=False, parallel_finetune=True)
  File "/home/andy/ws_nvidia/src/curobo/src/curobo/wrap/reacher/motion_gen.py", line 1870, in warmup
    self.graph_planner.warmup(
  File "/home/andy/ws_nvidia/src/curobo/src/curobo/graph/prm.py", line 520, in warmup
    self.build_graph(
  File "/home/andy/ws_nvidia/src/curobo/src/curobo/graph/prm.py", line 494, in build_graph
    v_set = self.get_biased_vertex_set(
  File "/home/andy/.local/share/ov/pkg/isaac-sim-2023.1.1/kit/python/lib/python3.10/contextlib.py", line 79, in inner
    return func(*args, **kwds)
  File "/home/andy/ws_nvidia/src/curobo/src/curobo/graph/graph_base.py", line 391, in get_biased_vertex_set
    x_search = self.get_feasible_sample_set(x_samples)
  File "/home/andy/ws_nvidia/src/curobo/src/curobo/graph/graph_base.py", line 266, in get_feasible_sample_set
    mask = self.mask_samples(x_samples)
  File "/home/andy/ws_nvidia/src/curobo/src/curobo/graph/graph_base.py", line 272, in mask_samples
    return self._mask_samples_cuda_graph(x_samples)
  File "/home/andy/.local/share/ov/pkg/isaac-sim-2023.1.1/kit/python/lib/python3.10/contextlib.py", line 79, in inner
    return func(*args, **kwds)
  File "/home/andy/ws_nvidia/src/curobo/src/curobo/graph/graph_base.py", line 288, in _mask_samples_cuda_graph
    feasible = self._cuda_graph_rollout_constraint(
  File "/home/andy/ws_nvidia/src/curobo/src/curobo/graph/graph_base.py", line 317, in _cuda_graph_rollout_constraint
    metrics = self.rollout_fn.rollout_constraint_cuda_graph(
  File "/home/andy/ws_nvidia/src/curobo/src/curobo/rollout/arm_base.py", line 558, in rollout_constraint_cuda_graph
    self._cu_rollout_constraint_out_metrics = self.constraint_fn(
  File "/home/andy/ws_nvidia/src/curobo/src/curobo/rollout/arm_base.py", line 401, in constraint_fn
    coll_constraint = self.primitive_collision_constraint.forward(
  File "/home/andy/ws_nvidia/src/curobo/src/curobo/rollout/cost/primitive_collision_cost.py", line 179, in discrete_fn
    dist = self.coll_check_fn(
  File "/home/andy/ws_nvidia/src/curobo/src/curobo/geom/sdf/world_mesh.py", line 483, in get_sphere_collision
    d = self._get_sdf(
  File "/home/andy/ws_nvidia/src/curobo/src/curobo/geom/sdf/world_mesh.py", line 361, in _get_sdf
    d = SdfMeshWarpPy.apply(
  File "/home/andy/.local/share/ov/pkg/isaac-sim-2023.1.1/extscache/omni.pip.torch-2_0_1-2.0.2+105.1.lx64/torch-2-0-1/torch/autograd/function.py", line 506, in apply
    return super().apply(*args, **kwargs)  # type: ignore[misc]
  File "/home/andy/ws_nvidia/src/curobo/src/curobo/geom/sdf/warp_primitives.py", line 538, in forward
    wp.launch(
  File "/home/andy/.local/share/ov/pkg/isaac-sim-2023.1.1/extscache/omni.warp.core-1.0.0-beta.2/warp/context.py", line 3358, in launch
    if not module.load(device):
  File "/home/andy/.local/share/ov/pkg/isaac-sim-2023.1.1/extscache/omni.warp.core-1.0.0-beta.2/warp/context.py", line 1388, in load
    builder = ModuleBuilder(self, self.options)
  File "/home/andy/.local/share/ov/pkg/isaac-sim-2023.1.1/extscache/omni.warp.core-1.0.0-beta.2/warp/context.py", line 1041, in __init__
    self.build_function(f)
  File "/home/andy/.local/share/ov/pkg/isaac-sim-2023.1.1/extscache/omni.warp.core-1.0.0-beta.2/warp/context.py", line 1087, in build_function
    func.adj.build(self)
  File "/home/andy/.local/share/ov/pkg/isaac-sim-2023.1.1/extscache/omni.warp.core-1.0.0-beta.2/warp/codegen.py", line 551, in build
    raise e
  File "/home/andy/.local/share/ov/pkg/isaac-sim-2023.1.1/extscache/omni.warp.core-1.0.0-beta.2/warp/codegen.py", line 542, in build
    adj.eval(adj.tree.body[0])
  File "/home/andy/.local/share/ov/pkg/isaac-sim-2023.1.1/extscache/omni.warp.core-1.0.0-beta.2/warp/codegen.py", line 1766, in eval
    return emit_node(adj, node)
  File "/home/andy/.local/share/ov/pkg/isaac-sim-2023.1.1/extscache/omni.warp.core-1.0.0-beta.2/warp/codegen.py", line 1054, in emit_FunctionDef
    adj.eval(f)
  File "/home/andy/.local/share/ov/pkg/isaac-sim-2023.1.1/extscache/omni.warp.core-1.0.0-beta.2/warp/codegen.py", line 1766, in eval
    return emit_node(adj, node)
  File "/home/andy/.local/share/ov/pkg/isaac-sim-2023.1.1/extscache/omni.warp.core-1.0.0-beta.2/warp/codegen.py", line 1633, in emit_Assign
    rhs = adj.eval(node.value)
  File "/home/andy/.local/share/ov/pkg/isaac-sim-2023.1.1/extscache/omni.warp.core-1.0.0-beta.2/warp/codegen.py", line 1766, in eval
    return emit_node(adj, node)
  File "/home/andy/.local/share/ov/pkg/isaac-sim-2023.1.1/extscache/omni.warp.core-1.0.0-beta.2/warp/codegen.py", line 1479, in emit_Call
    out = adj.add_call(func=func, args=args, kwds=kwds, templates=templates, min_outputs=min_outputs)
  File "/home/andy/.local/share/ov/pkg/isaac-sim-2023.1.1/extscache/omni.warp.core-1.0.0-beta.2/warp/codegen.py", line 779, in add_call
    raise Exception(
Exception: Error while parsing function "mesh_query_point_fn" at /home/andy/ws_nvidia/src/curobo/src/curobo/geom/sdf/warp_primitives.py:25:
    collide_result = wp.mesh_query_point(idx, point, max_distance)
Couldn't find function overload for 'mesh_query_point' that matched inputs with types: [uint64, vec3f, float32]
```

ybbng, 2024-06-18:
I also meet this problem yestoday,  but I already know how to solve it. Version 0.7.3 modified some code and added content related to warp.mesh_query_point in warp_primitives.py. It computes the closest point. You should switch the branch to isaac-3.0. I believe that the latest main branch, including the changes in version 0.7.3, has not yet been adapted to Isaac.

In short, you should:
1、git checkout  isaac-3.0
2、[omni_python -m pip install -e .[isaacsim] --no-build-isolation  ](https://curobo.org/get_started/1_install_instructions.html )
3  omni_python examples/isaac_sim/motion_gen_reacher.py

balakumar-s, 2024-06-18:
We recently updated to a newer version of warp that allows for more runtime kernel customization. This unfortunately broke isaac sim examples. We are working on getting our examples to work in isaac sim 4.0 which has the newest warp. Meanwhile, use an older version of curobo (v0.7.2 or lower).

balakumar-s, 2024-06-24:
We added a fallback version of the functions to work with older warp versions in the latest main commit. Re-open the issue if you find this to still be broken.

closed 2024-06-24 by balakumar-s