NVlabs/curobo #303: Issue with running example file.

Huangruopeng, 2024-06-19:
If it’s not a bug, please use discussions: https://github.com/NVlabs/curobo/discussions

Please provide the below information in addition to your issue:

1. cuRobo installation mode (choose from [python, isaac sim, docker python, docker isaac sim]):
2. python version:
3. Isaac Sim version (if using): 

------
1. Installed cuRobo for isaac sim from https://curobo.org/get_started/1_install_instructions.html
2. Python version is Python 3.10.12
3. Isaac Sim version is 2023.1.1

tried running the example file after a successful installation but on running this example `omni_python examples/isaac_sim/collision_checker_example.py`, isaac sim crashes on pressing play. Also tried the motion generation example with both franka and ur5e but the isaac sim window immediately crashes after launching. This shows up in the terminal with a Could not find category 'Replicator:Annotators' warning.

```
lm-2023@lm2023-XPS-8950:~/curobo$ omni_python examples/isaac_sim/motion_gen_reacher.py --robot ur5e.yml
Starting kit application with the following args:  ['/home/lm-2023/.local/share/ov/pkg/isaac-sim-2023.1.1/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py', '/home/lm-2023/.local/share/ov/pkg/isaac-sim-2023.1.1/apps/omni.isaac.sim.python.kit', '--/app/tokens/exe-path=/home/lm-2023/.local/share/ov/pkg/isaac-sim-2023.1.1/kit', '--/persistent/app/viewport/displayOptions=3094', '--/rtx/materialDb/syncLoads=True', '--/rtx/hydra/materialSyncLoads=True', '--/omni.kit.plugin/syncUsdLoads=True', '--/app/renderer/resolution/width=1920', '--/app/renderer/resolution/height=1080', '--/app/window/width=1440', '--/app/window/height=900', '--/renderer/multiGpu/enabled=True', '--/app/fastShutdown=True', '--ext-folder', '/home/lm-2023/.local/share/ov/pkg/isaac-sim-2023.1.1/exts', '--ext-folder', '/home/lm-2023/.local/share/ov/pkg/isaac-sim-2023.1.1/apps', '--/physics/cudaDevice=0', '--portable']
Passing the following args to the base kit application:  ['--robot', 'ur5e.yml']
[Info] [carb] Logging to file: /home/lm-2023/.local/share/ov/pkg/isaac-sim-2023.1.1/kit/logs/Kit/Isaac-Sim/2023.1/kit_20240619_141206.log
2024-06-19 21:12:06 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
[0.035s] [ext: omni.kit.async_engine-0.0.0] startup
[0.192s] [ext: omni.activity.core-1.0.1] startup
[0.195s] [ext: omni.assets.plugins-0.0.0] startup
[0.196s] [ext: omni.stats-0.0.0] startup
[0.196s] [ext: omni.client-1.0.1] startup
[0.201s] [ext: omni.activity.profiler-1.0.2] startup
[0.202s] [ext: omni.gpu_foundation-0.0.0] startup
[0.208s] [ext: omni.rtx.shadercache.vulkan-1.0.0] startup
[0.209s] [ext: carb.windowing.plugins-1.0.0] startup
[0.213s] [ext: omni.kit.renderer.init-0.0.0] startup

|---------------------------------------------------------------------------------------------|
| Driver Version: 545.29.06     | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 3080          | Yes: 0 |     | 10486   MB | 10de      | 0          |
|     |                                  |        |     |            | 2216      | a770c3ac.. |
|     |                                  |        |     |            | 1         |            |
|=============================================================================================|
| OS: 22.04.4 LTS (Jammy Jellyfish) ubuntu, Version: 22.04.4, Kernel: 6.5.0-35-generic
| XServer Vendor: The X.Org Foundation, XServer Version: 12101004 (1.21.1.4)
| Processor: 12th Gen Intel(R) Core(TM) i7-12700 | Cores: 12 | Logical: 24
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 31810 | Free Memory: 26057
| Total Page/Swap (MB): 2047 | Free Page/Swap: 785
|---------------------------------------------------------------------------------------------|
[0.554s] [ext: omni.kit.pipapi-0.0.0] startup
[0.555s] [ext: omni.kit.pip_archive-0.0.0] startup
[0.556s] [ext: omni.pip.compute-1.2.0] startup
[0.556s] [ext: omni.pip.cloud-1.0.1] startup
[0.558s] [ext: omni.isaac.core_archive-2.2.1] startup
[0.558s] [ext: omni.pip.torch-2_0_1-2.0.2] startup
[0.580s] [ext: omni.kit.telemetry-0.5.0] startup
[0.598s] [ext: omni.kit.loop-isaac-1.1.0] startup
[0.600s] [ext: omni.kit.test-0.0.0] startup
[0.601s] [ext: omni.isaac.ml_archive-1.1.3] startup
[0.601s] [ext: omni.mtlx-0.1.0] startup
[0.601s] [ext: omni.usd.config-1.0.3] startup
[0.606s] [ext: omni.gpucompute.plugins-0.0.0] startup
[0.606s] [ext: omni.usd.libs-1.0.0] startup
[0.678s] [ext: omni.appwindow-1.1.5] startup
[1.007s] [ext: omni.kit.renderer.core-0.0.0] startup
[2.426s] [ext: omni.kit.renderer.capture-0.0.0] startup
[2.428s] [ext: omni.kit.renderer.imgui-0.0.0] startup
[2.474s] [ext: carb.audio-0.1.0] startup
[2.476s] [ext: omni.ui-2.18.5] startup
[... 311 lines cut ...]
  File "/home/lm-2023/curobo/src/curobo/rollout/arm_base.py", line 324, in _init_after_config_load
    self.null_convergence = DistCost(self.convergence_cfg.null_space_cfg)
  File "/home/lm-2023/curobo/src/curobo/rollout/cost/dist_cost.py", line 343, in __init__
    self._l2_dof_kernel = make_l2_kernel(self.dof)
  File "/home/lm-2023/curobo/src/curobo/rollout/cost/dist_cost.py", line 227, in make_l2_kernel
    return wp.Kernel(forward_l2_loop_warp)
TypeError: Kernel.__init__() missing 2 required positional arguments: 'key' and 'module'
2024-06-19 21:12:15 [8,894ms] [Warning] [carb] [Plugin: omni.spectree.delegate.plugin] Module /home/lm-2023/.local/share/ov/pkg/isaac-sim-2023.1.1/kit/exts/omni.usd_resolver/bin/libomni.spectree.delegate.plugin.so remained loaded after unload request
2024-06-19 21:12:15 [8,896ms] [Warning] [omni.stageupdate.plugin] Deprecated: direct use of IStageUpdate callbacks is deprecated. Use IStageUpdate::getStageUpdate instead.
2024-06-19 21:12:15 [8,897ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-06-19 21:12:15 [8,897ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-06-19 21:12:15 [8,897ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-06-19 21:12:15 [8,897ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-06-19 21:12:15 [8,897ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-06-19 21:12:15 [8,897ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-06-19 21:12:15 [8,897ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-06-19 21:12:15 [8,897ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-06-19 21:12:15 [8,897ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-06-19 21:12:15 [8,897ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-06-19 21:12:15 [8,897ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-06-19 21:12:15 [8,897ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Core' for removal
2024-06-19 21:12:15 [8,899ms] [Warning] [carb.audio.context] 1 contexts were leaked
2024-06-19 21:12:15 [8,916ms] [Warning] [carb] Recursive unloadAllPlugins() detected!
2024-06-19 21:12:15 [8,924ms] [Warning] [omni.core.ITypeFactory] Module /home/lm-2023/.local/share/ov/pkg/isaac-sim-2023.1.1/kit/exts/omni.activity.core/bin/libomni.activity.core.plugin.so remained loaded after unload request.
```

JenniferBuehler, 2024-06-20:
I can confirm that I am experiencing the same. I'm on Ubuntu 22.04, python 3.10.12, and downloaded Isaac 2023.1.1 just yesterday from Omniverse. I get the same for 2023.0.1-hotfix as well.

For `examples/isaac_sim/motion_gen_reacher.py` I get what you do:
```
Traceback (most recent call last):
  File "/home/jenny/omni/curobo/examples/isaac_sim/motion_gen_reacher.py", line 437, in <module>
    main()
  File "/home/jenny/omni/curobo/examples/isaac_sim/motion_gen_reacher.py", line 213, in main
    motion_gen_config = MotionGenConfig.load_from_robot_config(
  File "/home/jenny/omni/curobo/src/curobo/wrap/reacher/motion_gen.py", line 642, in load_from_robot_config
    ik_solver_cfg = IKSolverConfig.load_from_robot_config(
  File "/home/jenny/.local/share/ov/pkg/isaac-sim-2023.1.1/kit/python/lib/python3.10/contextlib.py", line 79, in inner
    return func(*args, **kwds)
  File "/home/jenny/omni/curobo/src/curobo/wrap/reacher/ik_solver.py", line 316, in load_from_robot_config
    arm_rollout_mppi = ArmReacher(cfg)
  File "/home/jenny/.local/share/ov/pkg/isaac-sim-2023.1.1/kit/python/lib/python3.10/contextlib.py", line 79, in inner
    return func(*args, **kwds)
  File "/home/jenny/omni/curobo/src/curobo/rollout/arm_reacher.py", line 167, in __init__
    ArmBase.__init__(self)
  File "/home/jenny/.local/share/ov/pkg/isaac-sim-2023.1.1/kit/python/lib/python3.10/contextlib.py", line 79, in inner
    return func(*args, **kwds)
  File "/home/jenny/omni/curobo/src/curobo/rollout/arm_base.py", line 233, in __init__
    self._init_after_config_load()
  File "/home/jenny/.local/share/ov/pkg/isaac-sim-2023.1.1/kit/python/lib/python3.10/contextlib.py", line 79, in inner
    return func(*args, **kwds)
  File "/home/jenny/omni/curobo/src/curobo/rollout/arm_base.py", line 324, in _init_after_config_load
    self.null_convergence = DistCost(self.convergence_cfg.null_space_cfg)
  File "/home/jenny/omni/curobo/src/curobo/rollout/cost/dist_cost.py", line 343, in __init__
    self._l2_dof_kernel = make_l2_kernel(self.dof)
  File "/home/jenny/omni/curobo/src/curobo/rollout/cost/dist_cost.py", line 227, in make_l2_kernel
    return wp.Kernel(forward_l2_loop_warp)
TypeError: Kernel.__init__() missing 2 required positional arguments: 'key' and 'module'
``` 

And for `collision_checker.py` it's actually a different one (right after hitting play):

```
  File "/home/jenny/omni/curobo/examples/isaac_sim/collision_checker.py", line 205, in <module>
    main()
  File "/home/jenny/omni/curobo/examples/isaac_sim/collision_checker.py", line 179, in main
    d, d_vec = model.get_collision_vector(x_sph)
  File "/home/jenny/omni/curobo/src/curobo/wrap/model/robot_world.py", line 244, in get_collision_vector
    d = self.collision_cost.forward(x_sph, env_query_idx=env_query_idx)
  File "/home/jenny/omni/curobo/src/curobo/rollout/cost/primitive_collision_cost.py", line 179, in discrete_fn
    dist = self.coll_check_fn(
  File "/home/jenny/omni/curobo/src/curobo/geom/sdf/world_mesh.py", line 435, in get_sphere_distance
    d = self._get_sdf(
  File "/home/jenny/omni/curobo/src/curobo/geom/sdf/world_mesh.py", line 361, in _get_sdf
    d = SdfMeshWarpPy.apply(
  File "/home/jenny/.local/share/ov/pkg/isaac-sim-2023.1.1/extscache/omni.pip.torch-2_0_1-2.0.2+105.1.lx64/torch-2-0-1/torch/autograd/function.py", line 506, in apply
    return super().apply(*args, **kwargs)  # type: ignore[misc]
  File "/home/jenny/omni/curobo/src/curobo/geom/sdf/warp_primitives.py", line 538, in forward
    wp.launch(
  File "/home/jenny/.local/share/ov/pkg/isaac-sim-2023.1.1/extscache/omni.warp.core-1.0.0-beta.2/warp/context.py", line 3358, in launch
    if not module.load(device):
  File "/home/jenny/.local/share/ov/pkg/isaac-sim-2023.1.1/extscache/omni.warp.core-1.0.0-beta.2/warp/context.py", line 1388, in load
    builder = ModuleBuilder(self, self.options)
  File "/home/jenny/.local/share/ov/pkg/isaac-sim-2023.1.1/extscache/omni.warp.core-1.0.0-beta.2/warp/context.py", line 1041, in __init__
    self.build_function(f)
  File "/home/jenny/.local/share/ov/pkg/isaac-sim-2023.1.1/extscache/omni.warp.core-1.0.0-beta.2/warp/context.py", line 1087, in build_function
    func.adj.build(self)
  File "/home/jenny/.local/share/ov/pkg/isaac-sim-2023.1.1/extscache/omni.warp.core-1.0.0-beta.2/warp/codegen.py", line 551, in build
    raise e
  File "/home/jenny/.local/share/ov/pkg/isaac-sim-2023.1.1/extscache/omni.warp.core-1.0.0-beta.2/warp/codegen.py", line 542, in build
    adj.eval(adj.tree.body[0])
  File "/home/jenny/.local/share/ov/pkg/isaac-sim-2023.1.1/extscache/omni.warp.core-1.0.0-beta.2/warp/codegen.py", line 1766, in eval
    return emit_node(adj, node)
  File "/home/jenny/.local/share/ov/pkg/isaac-sim-2023.1.1/extscache/omni.warp.core-1.0.0-beta.2/warp/codegen.py", line 1054, in emit_FunctionDef
    adj.eval(f)
  File "/home/jenny/.local/share/ov/pkg/isaac-sim-2023.1.1/extscache/omni.warp.core-1.0.0-beta.2/warp/codegen.py", line 1766, in eval
    return emit_node(adj, node)
  File "/home/jenny/.local/share/ov/pkg/isaac-sim-2023.1.1/extscache/omni.warp.core-1.0.0-beta.2/warp/codegen.py", line 1633, in emit_Assign
    rhs = adj.eval(node.value)
  File "/home/jenny/.local/share/ov/pkg/isaac-sim-2023.1.1/extscache/omni.warp.core-1.0.0-beta.2/warp/codegen.py", line 1766, in eval
    return emit_node(adj, node)
  File "/home/jenny/.local/share/ov/pkg/isaac-sim-2023.1.1/extscache/omni.warp.core-1.0.0-beta.2/warp/codegen.py", line 1479, in emit_Call
[... cut ...]
omni-python pip install usd-core==23.11
```

Did you have to do that too?
If I would guess (I'm new to this code base so it's just a guess really), that sounds like it could potentially relate to the second error, at least it's calling a file that contains the word `core` ;)

I have unfortunately only very limited time this week and I will probably have to move past the examples, but if I hit the same issues in my other use case, I will report here that it is an issue beyond just the examples (and the fix if I find it).

JenniferBuehler, 2024-06-20:
Confirm that this is not limited to the examples, at least the error you also reported.

It happens in `load_from_rorot_config` but ONLY if the SimulationApp instance has previously been created. Very simple code to reproduce:

```python
# isaac
# -----------

# >>>>>>>>>> Set this to True to enable the crash!!!
load_sim = False

if load_sim:
    from omni.isaac.kit import SimulationApp
    headless = True
    simulation_app = SimulationApp(
        {
            "headless": headless,
            "width": "1920",
            "height": "1080",
        }
    )
        
# cuRobo
# -----------
from curobo.wrap.reacher.motion_gen import MotionGenConfig
from curobo.util_file import load_yaml

# in this simple example, I just have a config in the current working directory
robot_cfg = load_yaml("<path-to-any-yaml>")["robot_cfg"]
robot_cfg["kinematics"]["external_asset_path"] = "." 
            
motion_gen_config = MotionGenConfig.load_from_robot_config(
    robot_cfg
)   
```

JenniferBuehler, 2024-06-20:
I had a closer look and NVidia warp is installed twice, and the `Kernel` class is different in these two. 

```
HOME/.local/share/ov/pkg/isaac-sim-2023.1.1/extscache/omni.warp.core-1.0.0-beta.2/warp/context.py
HOME/.local/share/ov/pkg/isaac-sim-2023.1.1/kit/python/lib/python3.10/site-packages/warp/context.py
```

It's the one in `extscache` that gets used, as it expects `key` and `module` arguments. That's the file installed by IsaacSim:

```python
# caches source and compiled entry points for a kernel (will be populated after module loads)
class Kernel:
    def __init__(self, func, key, module, options=None, code_transformers=[]):
        self.func = func
```

And the one that would be expected by curobo is the other one:

```python
# caches source and compiled entry points for a kernel (will be populated after module loads)
class Kernel:
    def __init__(self, func, key=None, module=None, options=None, code_transformers=None):
        self.func = func
```

The first one, which causes the issue, seems odd to me in general (but I'm also absolutely new to this codebase). Why? In the [original repo](https://github.com/NVIDIA/warp/blob/release-1.0.0-beta.5/warp/context.py) for release 1.0.0 beta, the parameters have default values too. Therefore, to test what's up with this, I just removed the whole package (I know, it might fly in my face at another occasion):

```
mv HOME/.local/share/ov/pkg/isaac-sim-2023.1.1/extscache/omni.warp* ~/isaacbackups
```

I kept a backup of course, just in case I need to move it back. And: no more exception. It also solves the other exception in `collision_checker.py` that I reported above.

The `motion_gen_reacher.py` still doesn't load the robot visuals, but that's an entirely different problem. When I move the cube I can see in the XForms that the arm technically is following the cube, so the curobo part seems to be working. The collision checker works just as expected.

I will report this in the isaac forum as well, maybe someone there knows what's up with this warp beta version. Maybe it's a fork from the original which was modified.

P.S. the duplicated warp package has nothing to do with the installation of `usd-core` as mentioned above (confirm with `omni-python -m pip show usd-core -f`)

chiawenwang, 2024-06-24:
> I had a closer look and NVidia warp is installed twice, and the `Kernel` class is different in these two.
>
> ``` [...]

I just follow your advice by：
`mv HOME/.local/share/ov/pkg/isaac-sim-2023.1.1/extscache/omni.warp* ~/isaacbackups`

But it still doesn't work.

I think this problem is link to the version of Isaac sim? and my version is 2023.1.1 in Ubuntu20.04. I also try to add --robot ur5e.yml. but not work too.

do you have more process on this issue?

JenniferBuehler, 2024-06-24:
Silly question, but I guess you added the `$` to the HOME env that I forgot to add in the text above ;)

Can you do a `sudo updatedb && locate context.py | grep warp` and see if there are several context.py files (which is where the `Kernel` class lives)?

Make sure that you use the `python.py` from your isaac package as well, which sets the python envorinment variables. Having several isaac versions is not a problem then, I have several installed.

balakumar-s, 2024-06-24:
We fixed this issue in the latest main commit. cuRobo now also works with older warp versions. Please re-open if it still doesn't work for you.

closed 2024-06-24 by balakumar-s

JenniferBuehler, 2024-06-25:
Great news, thank you!