NVlabs/curobo #306: Using function "update_locked_joints" does not change motion_gen.kinematics.kinematics_config.lock_jointstate

KIKI007, 2024-06-21:
1. cuRobo installation mode: python
2. python version: 3.8
------
Issue Details

Problem
Using function "update_locked_joints" does not change motion_gen.kinematics.kinematics_config.lock_jointstate

In function "update_locked_joints" from motion_gen.py
``` python
robot_config_dict["kinematics"]["lock_joints"] = lock_joints
robot_cfg = RobotConfig.from_dict(robot_config_dict, self.tensor_args)
self.kinematics.update_kinematics_config(robot_cfg.kinematics.kinematics_config)
```
However,  the function "update_kinematics_config" currently does not update its **lock_jointstate**
Can you help me with this issue?

balakumar-s, 2024-07-01:
Good catch, we directly updated only the GPU tensors while lock_joint_state is on cpu. We will fix this in a future release.

balakumar-s, 2024-07-22:
Fixed in latest main.

closed 2024-07-22 by balakumar-s