NVlabs/curobo #310: curobo_execution_error

LnZeeyuan, 2024-06-24:
Hardware : NV Geforce 24G 
Software: Ubuntu22.04  CUDA 11.8 PyTorch2.2.1 Python3.10 isaac-sim-2023.1.1/saac-sim-2023.1.0-hotfix.1

issues: We already installed curobo, according to "https://curobo.org/get_started/1_install_instructions.html".
        When executing sh"omni_python examples/isaac_sim/batch_motion_gen_reacher.py", an error occurs

problem present :
	Traceback (most recent call last):
	File "/home/hiteam/.local/share/ov/pkg/isaac_sim-2023.1.1/curobo/examples/isaac_sim/motion_gen_reacher.py", line 437, in <module>
		main()
	File "/home/hiteam/.local/share/ov/pkg/isaac_sim-2023.1.1/curobo/examples/isaac_sim/motion_gen_reacher.py", line 213, in main
		motion_gen_config = MotionGenConfig.load_from_robot_config(
	File "/home/hiteam/.local/share/ov/pkg/isaac_sim-2023.1.1/curobo/src/curobo/wrap/reacher/motion_gen.py", line 642, in load_from_robot_config
		ik_solver_cfg = IKSolverConfig.load_from_robot_config(
	File "/home/hiteam/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/python/lib/python3.10/contextlib.py", line 79, in inner
		return func(*args, **kwds)
	File "/home/hiteam/.local/share/ov/pkg/isaac_sim-2023.1.1/curobo/src/curobo/wrap/reacher/ik_solver.py", line 316, in load_from_robot_config
		arm_rollout_mppi = ArmReacher(cfg)
	File "/home/hiteam/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/python/lib/python3.10/contextlib.py", line 79, in inner
		return func(*args, **kwds)
	File "/home/hiteam/.local/share/ov/pkg/isaac_sim-2023.1.1/curobo/src/curobo/rollout/arm_reacher.py", line 167, in __init__
		ArmBase.__init__(self)
	File "/home/hiteam/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/python/lib/python3.10/contextlib.py", line 79, in inner
		return func(*args, **kwds)
	File "/home/hiteam/.local/share/ov/pkg/isaac_sim-2023.1.1/curobo/src/curobo/rollout/arm_base.py", line 233, in __init__
		self._init_after_config_load()
	File "/home/hiteam/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/python/lib/python3.10/contextlib.py", line 79, in inner
		return func(*args, **kwds)
	File "/home/hiteam/.local/share/ov/pkg/isaac_sim-2023.1.1/curobo/src/curobo/rollout/arm_base.py", line 324, in _init_after_config_load
		self.null_convergence = DistCost(self.convergence_cfg.null_space_cfg)
	File "/home/hiteam/.local/share/ov/pkg/isaac_sim-2023.1.1/curobo/src/curobo/rollout/cost/dist_cost.py", line 343, in __init__
		self._l2_dof_kernel = make_l2_kernel(self.dof)
	File "/home/hiteam/.local/share/ov/pkg/isaac_sim-2023.1.1/curobo/src/curobo/rollout/cost/dist_cost.py", line 227, in make_l2_kernel
		return wp.Kernel(forward_l2_loop_warp)
	TypeError: Kernel.__init__() missing 2 required positional arguments: 'key' and 'module'

LnZeeyuan, 2024-06-24:
Another error

[image]

balakumar-s, 2024-06-24:
We pushed a fix for this, can you try the latest commit?

closed 2024-07-22 by balakumar-s