NVlabs/curobo #318: constrained_reacher example: wrong solution

curieuxjy, 2024-07-01:
1. cuRobo installation mode (choose from [python, isaac sim, docker python, docker isaac sim]): isaac sim
2. python version: use omni_python in isaac-sim 2023.1.1
3. Isaac Sim version (if using): 2023.1.1

------
Issue Details

When running the `isaac_sim/constrained_reacher.py` example, the motion generation is weird.
I attached the video showing that situation.

[Problem Screenshot](https://github.com/NVlabs/curobo/assets/40867411/42774f77-bd1c-4e8e-baf9-5485cf7ccdd6)

I'm debugging now and found that the motion planning result, cmd_plan, was `None`.
I did not modify anything from the original example code.

balakumar-s, 2024-07-01:
This is because we are importing Franka Panda from an urdf,  that doesn't work well in Isaac Sim>2023.1.0. You can try to run examples/isaac_sim/motion_gen_reacher.py and in that as well Franka Panda will not be stable.

arhanjain, 2024-07-02:
> This is because we are importing Franka Panda from an urdf, that doesn't work well in Isaac Sim>2023.1.0. You can try to run examples/isaac_sim/motion_gen_reacher.py and in that as well Franka Panda will not be stable.

Do you have a recommendation for a workaround for using a Panda with curobo in Isaac Sim>2023.1.0? As far as I know, the Omniverse now only allows IsaacSim 2023.1.0, 2023.1.1, and 4.0.0 to be installed.

curieuxjy, 2024-07-08:
It worked when I changed version 2023.1.1 to isaac-sim-2023.1.0-hotfix.1
@arhanjain @balakumar-s 

[Screencast from 2024년 07월 08일 16시 07분 05초.webm](https://github.com/NVlabs/curobo/assets/40867411/2bcfedeb-d1d1-43a1-b0c9-77f2791ab577)

MomoHsieh, 2024-07-17:
Do you have plan to support hold_vec_weight not only along x, y or z but also randomly vector?

balakumar-s, 2024-07-22:
We have updated the examples to work with Isaac Sim 4.0 including Franka Panda.

> Do you have plan to support hold_vec_weight not only along x, y or z but also randomly vector?

We do not have this in our roadmap. However, you can look at the code and make small changes to make it work for arbitrary axis. Create a discussion if you want some pointers @MomoHsieh.

closed 2024-07-22 by balakumar-s