NVlabs/curobo #321: frank.yml is broken

MikeWise2718, 2024-07-03:
Most of the examples (and documentation) - like `motion_gen_reacher.py` use the `franka.yml` robot configuration as default, which is broken somehow. The robot does not ever stop moving after an initial push. I messed around with the stiffness and damping in the joints - which is where the error looks to be - but I couldn't get it to have an effect. So I don't know what is wrong or maybe how it works.

I did however notice that the `ur5e.yml` robot works just fine.

https://github.com/NVlabs/curobo/assets/7562887/0ffa7abd-85ee-448e-a387-0fbe3298e4e5

raymondyu5, 2024-07-03:
I think this is a common problem faced by people with newer versions of IsaacSim such as 2023.1.1+ where the urdf is being loaded weirdly.

Not sure what to really do since I am also trying to use cuRobo code for newer versions of IsaacSim. Would be nice to have a fix to this especially since you can now no longer download older versions of Isaac.

MikeWise2718, 2024-07-03:
I thinkmimgot the same behavior on 2023.1.0 (hotfix). I'll see if I can get a video of it tomorrow.

pmj110119, 2024-07-08:
Urgently hope that this can be fixed in the new version of Isaac Sim!

Sajjan2406, 2024-07-12:
Even i faced the same problem. If you notice, initially the end effector is in collission(before the simulation). When the play button is pressed, due to the collisison the robot misbehaves. But if you change the robot(e.g., ur10e), it works fine

balakumar-s, 2024-07-12:
We have a fix incoming by next Wednesday for this. This fix also brings support for Isaac sim 4.0.0.

balakumar-s, 2024-07-22:
Fixed in latest main.

closed 2024-07-22 by balakumar-s