NVlabs/curobo #333: `FINETUNE_TRAJOPT_FAIL` for a customized robot arm with a torso

HendaIman, 2024-07-12:
Please provide the below information in addition to your issue:

1. cuRobo installation mode (choose from [python, isaac sim, docker python, docker isaac sim]): isaac_sim
2. python version: 3.10
3. Isaac Sim version (if using): 2023.1.0-hotfix.1

------
I'm using cuRobo motion generation for a customized robot. The robot has a 6-DoF arm, mounted on a torso with 4 revolute joints. I followed the tutorial about how to add a new robot and successfully get the motion generation work for the 6-DoF arm. However, I couldn't get it work for the same arm plus the torso. The returned `MotionGenStatus` is `FINETUNE_TRAJOPT_FAIL`. The IK example works though. The motion generation works if I lock all torso joints. What's even more strange is, the motion gen doesn't work for the torso itself (ee link is the upper body). I guess this might suggest there is something wrong with my robot torso model? Appreciate any suggestions about how to debug this.

A summary of when it works and not:

- 6-DoF arm. IK: Yes. MotionGen: Yes.
- 6-DoF arm + torso (locked joints). IK: Yes. MotionGen: Yes.
- 6-DoF arm + 4-DoF torso. IK: Yes. MotionGen: No, always `FINETUNE_TRAJOPT_FAIL`.
- 4-DoF torso. IK: No. MotionGen: No.

balakumar-s, 2024-07-12:
Finetune trajopt fail can happen if the trajectory is not within velocity, acceleration or jerk limits. Are the limits for the base lower than the arm?

HendaIman, 2024-07-13:
> Finetune trajopt fail can happen if the trajectory is not within velocity, acceleration or jerk limits. Are the limits for the base lower than the arm?

Thanks! Tuning them fixed the issue.

closed 2024-07-13 by HendaIman