NVlabs/curobo #347: Broken `motion_gen_reacher.py` script with the argument `--reactive`

yunfanjiang, 2024-07-28:
If it’s not a bug, please use discussions: https://github.com/NVlabs/curobo/discussions

Please provide the below information in addition to your issue:

1. cuRobo installation mode (choose from [python, isaac sim, docker python, docker isaac sim]): isaac sim
2. python version: 3.10
3. Isaac Sim version (if using): 4.0.0

------
Thanks for open-sourcing this project! When I tried to run the `moiton_gen_reacher.py` example with the `--reactive` argument, I got the following error.
```
Traceback (most recent call last):
  File "/home/yunfanj/workspace/curobo/examples/isaac_sim/motion_gen_reacher.py", line 437, in <module>
    main()
  File "/home/yunfanj/workspace/curobo/examples/isaac_sim/motion_gen_reacher.py", line 229, in main
    motion_gen.warmup(enable_graph=True, warmup_js_trajopt=False, parallel_finetune=True)
  File "/home/yunfanj/workspace/curobo_isaacsim4/src/curobo/wrap/reacher/motion_gen.py", line 1892, in warmup
    self.plan_single(
  File "/home/yunfanj/workspace/curobo_isaacsim4/src/curobo/wrap/reacher/motion_gen.py", line 1533, in plan_single
    result = self._plan_attempts(
  File "/home/yunfanj/workspace/curobo_isaacsim4/src/curobo/wrap/reacher/motion_gen.py", line 2998, in _plan_attempts
    result = self._plan_from_solve_state(
  File "/home/yunfanj/workspace/curobo_isaacsim4/src/curobo/wrap/reacher/motion_gen.py", line 3474, in _plan_from_solve_state
    opt_dt = torch.min(opt_dt[traj_result.success])
IndexError: too many indices for tensor of dimension 0
```

It seems that the error was caused by setting `optimize_dt = False` ([this line](https://github.com/NVlabs/curobo/blob/3690d28c54ee4ba23e2d7cc03a89a5845e4f9546/examples/isaac_sim/motion_gen_reacher.py#L215)). If I kept it True the reactive example works qualitatively well.

scharalambous-sony, 2024-08-05:
This is a bug. Give my fork of curobo a try, I believe it resolves this issue

balakumar-s, 2024-08-19:
This has been fixed in the latest commit.

closed 2024-08-19 by balakumar-s