NVlabs/curobo #348: nvblox issue with isaac_sim_4.0.0 docker

shengyangzhuang, 2024-07-29:
If it’s not a bug, please use discussions: https://github.com/NVlabs/curobo/discussions

Please provide the below information in addition to your issue:

1. cuRobo installation mode (choose from [python, isaac sim, docker python, docker isaac sim]): docker Isaac sim
2. python version: 3.8.10 (the version in the docker)
3. Isaac Sim version (if using): 4.0.0

------
Issue Details

realsense_viewer.py can run without any error. However, realsense_reacher.py example script will return cv2 window related error **if I enable showing the nvblox graph.**
```
**** Click Play to start simulation *****
Traceback (most recent call last):
  File "/pkgs/curobo/examples/isaac_sim/realsense_reacher.py", line 342, in <module>
    cv2.namedWindow("NVBLOX Example", cv2.WINDOW_NORMAL)
cv2.error: OpenCV(4.9.0) /io/opencv/modules/highgui/src/window.cpp:1255: error: (-2:Unspecified error) The function is not implemented. Rebuild the library with Windows, GTK+ 2.x or Cocoa support. If you are on Ubuntu or Debian, install libgtk2.0-dev and pkg-config, then re-run cmake or configure script in function 'cvNamedWindow'

2024-07-29 14:18:03 [154,456ms] [Warning] [carb] [Plugin: omni.sensors.tiled.plugin] Module /isaac-sim/extscache/omni.sensors.tiled-0.0.3+106.0.0.lx64.r/bin/libomni.sensors.tiled.plugin.so remained loaded after unload request
2024-07-29 14:18:03 [154,460ms] [Warning] [carb] [Plugin: omni.spectree.delegate.plugin] Module /isaac-sim/kit/exts/omni.usd_resolver/bin/libomni.spectree.delegate.plugin.so remained loaded after unload request
2024-07-29 14:18:03 [154,463ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-07-29 14:18:03 [154,463ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-07-29 14:18:03 [154,463ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-07-29 14:18:03 [154,463ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-07-29 14:18:03 [154,463ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-07-29 14:18:03 [154,463ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-07-29 14:18:03 [154,463ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-07-29 14:18:03 [154,463ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-07-29 14:18:03 [154,463ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-07-29 14:18:03 [154,463ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-07-29 14:18:03 [154,463ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-07-29 14:18:03 [154,463ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Core' for removal
2024-07-29 14:18:03 [154,512ms] [Warning] [carb] Recursive unloadAllPlugins() detected!

```
I didn't make any changes. I checked the original code in both scripts. They handle the cv2 window in a similar way. And I ran both scripts with omni_python. This is weird. Any comments are highly appreciated.

balakumar-s, 2024-07-31:
Can you try `python.sh -m pip install opencv-python` ?

xiayip, 2024-08-05:
just move 
`from omni.isaac.kit import SimulationApp`
after 
`import cv2`
solve the problem

balakumar-s, 2024-08-05:
Thanks for finding a fix. We will add this to the next release.

closed 2024-08-05 by balakumar-s