NVlabs/curobo #352: Encounter Runtime Error CUDA  error invalid argument while  trying "Motion Generation" example

linhy97, 2024-07-31:
Hi, i encounter the bug "RuntimeError: CUDA error: invalid argument' while trying the example "Motion Generation" in webpage https://curobo.org/get_started/2a_python_examples.html. Hope for your help~

1.Basic Environments:  
Ubuntu 22.04, Cuda11.8, Cudnn 8.9, python=3.10.12 pytorch=2.0.1+cu118, curobo=0.7.4.post1.dev0+dirty
I try torch 2.0.0, 2.2, also  failed and same  error.

2. The code i try is the example supplied in "Using in Python" Page.
```
# Third Party
import torch

# cuRobo
from curobo.types.math import Pose
from curobo.types.robot import JointState
from curobo.wrap.reacher.motion_gen import MotionGen, MotionGenConfig, MotionGenPlanConfig

world_config = {
    "mesh": {
        "base_scene": {
            "pose": [10.5, 0.080, 1.6, 0.043, -0.471, 0.284, 0.834],
            "file_path": "scene/nvblox/srl_ur10_bins.obj",
        },
    },
    "cuboid": {
        "table": {
            "dims": [5.0, 5.0, 0.2],  # x, y, z
            "pose": [0.0, 0.0, -0.1, 1, 0, 0, 0.0],  # x, y, z, qw, qx, qy, qz
        },
    },
}

motion_gen_config = MotionGenConfig.load_from_robot_config(
    "ur5e.yml",
    world_config,
    interpolation_dt=0.01,
)
motion_gen = MotionGen(motion_gen_config)
motion_gen.warmup()

retract_cfg = motion_gen.get_retract_config()

state = motion_gen.rollout_fn.compute_kinematics(
    JointState.from_position(retract_cfg.view(1, -1))
)

goal_pose = Pose.from_list([-0.4, 0.0, 0.4, 1.0, 0.0, 0.0, 0.0])  # x, y, z, qw, qx, qy, qz
start_state = JointState.from_position(
    torch.zeros(1, 6).cuda(),
    joint_names=[
        "shoulder_pan_joint",
        "shoulder_lift_joint",
        "elbow_joint",
        "wrist_1_joint",
        "wrist_2_joint",
        "wrist_3_joint",
    ],
)

result = motion_gen.plan_single(start_state, goal_pose, MotionGenPlanConfig(max_attempts=1))
traj = result.get_interpolated_plan()  # result.interpolation_dt has the dt between timesteps
print("Trajectory Generated: ", result.success)
```

4. Detailed Error Info:

{
	"name": "RuntimeError",
	"message": "CUDA error: invalid argument
CUDA kernel errors might be asynchronously reported at some other API call, so the stacktrace below might be incorrect.
For debugging consider passing CUDA_LAUNCH_BLOCKING=1.
[... 231 lines cut ...]
---> 58     R = lbfgs_step_cu.forward(
     59         step_vec,  # .view(-1),
     60         rho_buffer,  # .view(-1),
     61         y_buffer,  # .view(-1),
     62         s_buffer,  # .view(-1),
     63         q,
     64         grad_q,  # .view(-1),
     65         x_0,
     66         grad_0,
     67         epsilon,
     68         b,
     69         m,
     70         v_dim,
     71         stable_mode,
     72         use_shared_buffers,
     73     )
     74     step_v = R[0].view(step_vec.shape)
     76     # ctx.save_for_backward(batch_spheres, robot_spheres, link_mats, link_sphere_map)

RuntimeError: CUDA error: invalid argument
CUDA kernel errors might be asynchronously reported at some other API call, so the stacktrace below might be incorrect.
For debugging consider passing CUDA_LAUNCH_BLOCKING=1.
Compile with `TORCH_USE_CUDA_DSA` to enable device-side assertions.
"
}

balakumar-s, 2024-07-31:
Did you install git lfs? My guess is that the mesh file was not cloned. Try `sudo apt install git-lfs` followed by `cd curobo && git lfs pull`.

If that still doesn't work, does the below example work?

```
# Third Party
import torch

# cuRobo
from curobo.types.math import Pose
from curobo.types.robot import JointState
from curobo.wrap.reacher.motion_gen import MotionGen, MotionGenConfig, MotionGenPlanConfig

world_config = {
    "cuboid": {
        "table": {
            "dims": [5.0, 5.0, 0.2],  # x, y, z
            "pose": [0.0, 0.0, -0.1, 1, 0, 0, 0.0],  # x, y, z, qw, qx, qy, qz
        },
    },
}

motion_gen_config = MotionGenConfig.load_from_robot_config(
    "ur5e.yml",
    world_config,
    interpolation_dt=0.01,
)
motion_gen = MotionGen(motion_gen_config)
motion_gen.warmup()

retract_cfg = motion_gen.get_retract_config()

state = motion_gen.rollout_fn.compute_kinematics(
    JointState.from_position(retract_cfg.view(1, -1))
)

goal_pose = Pose.from_list([-0.4, 0.0, 0.4, 1.0, 0.0, 0.0, 0.0])  # x, y, z, qw, qx, qy, qz
start_state = JointState.from_position(
    torch.zeros(1, 6).cuda(),
    joint_names=[
        "shoulder_pan_joint",
        "shoulder_lift_joint",
        "elbow_joint",
        "wrist_1_joint",
        "wrist_2_joint",
        "wrist_3_joint",
    ],
)

result = motion_gen.plan_single(start_state, goal_pose, MotionGenPlanConfig(max_attempts=1))
traj = result.get_interpolated_plan()  # result.interpolation_dt has the dt between timesteps
print("Trajectory Generated: ", result.success)
```

linhy97, 2024-08-01:
I check the `git-lfs` and it was installed.
And then i try the code you supply, however it also doesn't work.
The error is the same while execute `motion_gen.warmup()`
It seems not the error of "srl_ur10_bins.obj" (87.2MB)

Error code detail:
```
{
	"name": "RuntimeError",
	"message": "CUDA error: invalid argument
CUDA kernel errors might be asynchronously reported at some other API call, so the stacktrace below might be incorrect.
For debugging consider passing CUDA_LAUNCH_BLOCKING=1.
Compile with `TORCH_USE_CUDA_DSA` to enable device-side assertions.
",
	"stack": "---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
Cell In[1], line 24
     18 motion_gen_config = MotionGenConfig.load_from_robot_config(
     19     \"ur5e.yml\",
     20     world_config,
     21     interpolation_dt=0.01,
     22 )
     23 motion_gen = MotionGen(motion_gen_config)
---> 24 motion_gen.warmup()
     26 retract_cfg = motion_gen.get_retract_config()
     28 state = motion_gen.rollout_fn.compute_kinematics(
     29     JointState.from_position(retract_cfg.view(1, -1))
     30 )

File ~/桌面/lhy/code/curobo/src/curobo/wrap/reacher/motion_gen.py:1863, in MotionGen.warmup(self, enable_graph, batch, warmup_js_trajopt, batch_env_mode, parallel_finetune, n_goalset, warmup_joint_index, warmup_joint_delta)
   1861     goal_state.position[..., warmup_joint_index] += warmup_joint_delta
   1862     for _ in range(3):
-> 1863         self.plan_single_js(
   1864             start_state.clone(),
   1865             goal_state.clone(),
   1866             MotionGenPlanConfig(max_attempts=1, enable_finetune_trajopt=True),
   1867         )
   1869 if enable_graph:
   1870     start_state = JointState.from_position(
   1871         self.rollout_fn.dynamics_model.retract_config.view(1, -1).clone(),
   1872         joint_names=self.rollout_fn.joint_names,
   1873     )

File ~/桌面/lhy/code/curobo/src/curobo/wrap/reacher/motion_gen.py:2073, in MotionGen.plan_single_js(self, start_state, goal_state, plan_config)
   2070         return result
   2072 for n in range(plan_config.max_attempts):
-> 2073     result = self._plan_js_from_solve_state(
   2074         solve_state, start_state, goal_state, plan_config=plan_config
   2075     )
   2076     time_dict[\"trajopt_time\"] += result.trajopt_time
   2077     time_dict[\"graph_time\"] += result.graph_time

File ~/桌面/lhy/code/curobo/src/curobo/wrap/reacher/motion_gen.py:3734, in MotionGen._plan_js_from_solve_state(self, solve_state, start_state, goal_state, plan_config)
   3732 if self.optimize_dt:
   3733     self.finetune_js_trajopt_solver.update_solver_dt(scaled_dt.item())
-> 3734 traj_result = self._solve_trajopt_from_solve_state(
   3735     goal,
   3736     solve_state,
   3737     seed_traj,
   3738     trajopt_instance=self.finetune_js_trajopt_solver,
   3739     num_seeds_override=solve_state.num_trajopt_seeds,
   3740     newton_iters=newton_iters,
   3741     return_all_solutions=False,
   3742 )
   3744 finetune_time += traj_result.solve_time
   3745 if torch.count_nonzero(traj_result.success) > 0 or not self.optimize_dt:

File /usr/lib/python3.10/contextlib.py:79, in ContextDecorator.__call__.<locals>.inner(*args, **kwds)
     76 @wraps(func)
     77 def inner(*args, **kwds):
[... 174 lines cut ...]
     59         step_vec,  # .view(-1),
     60         rho_buffer,  # .view(-1),
     61         y_buffer,  # .view(-1),
     62         s_buffer,  # .view(-1),
     63         q,
     64         grad_q,  # .view(-1),
     65         x_0,
     66         grad_0,
     67         epsilon,
     68         b,
     69         m,
     70         v_dim,
     71         stable_mode,
     72         use_shared_buffers,
     73     )
     74     step_v = R[0].view(step_vec.shape)
     76     # ctx.save_for_backward(batch_spheres, robot_spheres, link_mats, link_sphere_map)

RuntimeError: CUDA error: invalid argument
CUDA kernel errors might be asynchronously reported at some other API call, so the stacktrace below might be incorrect.
For debugging consider passing CUDA_LAUNCH_BLOCKING=1.
Compile with `TORCH_USE_CUDA_DSA` to enable device-side assertions.
"
}
```

balakumar-s, 2024-08-09:
Is it possible for you to run this inside a docker? That would help us reproduce the issue on our end.

hesic73, 2025-02-13:
I met with the same issue on a cluster.

balakumar-s, 2025-04-25:
We improved error handling for the CUDA kernels, hoping this resolves this issue. I am closing for now as we don't have a way to reproduce. Please re-open if issue persists and we can figure out a setup to reproduce.

closed 2025-04-25 by balakumar-s