NVlabs/curobo #359: Does cuRobo support robot with prismatic joint?

Shimadawallace, 2024-08-08:
Hi, I want to apply cuRobo's motion generation function to Hello Robot (https://hello-robot.com/) which has several prismatic joints, so I changed one of the revolute joint of ure5 to prismatic joint to see whether cuRobo supports primatic joint or not.
I found that after the changing, the ure5 robot in isaac sim don't move any more.
Could you tell me is it possible to apply cuRobo on a robot with prismatic joint?

Shimadawallace, 2024-08-09:
it's working with Panda robot with prismatic joint now

closed 2024-08-09 by Shimadawallace

balakumar-s, 2024-08-09:
Also see here for an example yml configuration for the stretch robot: https://github.com/NVlabs/curobo/discussions/146#discussioncomment-8402647