NVlabs/curobo #371: UR5 robot does not work motion_gen_reacher.py

Dhanushvarma, 2024-08-16:
If it’s not a bug, please use discussions: https://github.com/NVlabs/curobo/discussions

Please provide the below information in addition to your issue:

1. cuRobo installation mode (choose from [python, isaac sim, docker python, docker isaac sim]): isaac sim
2. python version: Python 3.10.14
3. Isaac Sim version (if using):  isaac-sim-4.0.0

------
Issue Details :

After following the [tutorial](https://curobo.org/tutorials/1_robot_configuration.html) for configuring a new robot, when trying to run the motion_gen_reacher.py script, the robot does not move to the target position (red cube), even after pressing the play button on Isaac Sim. I have noticed that this only happens for the ur5e.yml

 
[image]

balakumar-s, 2024-08-19:
Did you try moving the cube? If yes, can you share what's printed in the terminal? 

When planning fails, the status will be printed in the terminal starting with ``Plan did not converge to a solution:..``

aCodeDog, 2024-09-24:
same problem...

theo-bloesch, 2024-10-02:
I have the same problem too. Does anyone have found a solution ?

JemuelStanley47, 2024-10-16:
I have the same issue with Isaac sim 4.0.0 and cu11.8. Has anyone figured this out yet?

JemuelStanley47, 2024-11-11:
According to the [official solution](https://forums.developer.nvidia.com/t/curobo-ur5e-example/308613/3), the joint velocities at initial configuration is noisy for UR5e and it causes the planner to never be called, and hence requires adjusting the joint gains. Change the line (probably line 365) from

`if (np.max(np.abs(sim_js.velocities)) < 0.2) or args.reactive:`

to 

`if (np.max(np.abs(sim_js.velocities)) < 0.5) or args.reactive:`

I can acknowledge that it works.

theo-bloesch, 2024-11-20:
Thank you yes it worked !!!