NVlabs/curobo #373: remove_obstacle() function seems to fail work

linhy97, 2024-08-20:
Hello, when i try to use remove_obstacle() to update the world model, since i need to consider attach an object to the robot, and set the object invisible.

A simple example is listed below:

```
obstacle_table = Cuboid(name="obs_table", pose=[0, -0.35, -0.15, 1, 0, 0, 0], dims=[0.85, 1.2, 0.78])
list_obstacle_cube = [obstacle_table]
world_model_base = WorldConfig(cuboid=list_obstacle_cube)
print("Number of cuboids:", len(world_model_base.cuboid))

world_model_base.remove_obstacle(obstacle_table)
print("Number of cuboids after remove:", len(world_model_base.cuboid))
world_model_base.remove_obstacle('obs_table')
print("Number of cuboids after remove:", len(world_model_base.cuboid))
```

The corresponding output is:
```
Number of cuboids: 1
Number of cuboids after remove: 1
Number of cuboids after remove: 1
```

I am confused about the results after executing remove_obstacle() function. If the obstacle "obs_table" is removed, the number will change to zero?
Hope for your help~

will-44, 2024-11-22:
Hello, I have the same problem. Did you manage to solve it?

balakumar-s, 2024-11-22:
Our current implementation of collision checking only accesses `WorldConfig.objects` to parse the obstacles. The `remove_obstacle()` currently only removes it from `objects`. It doesn't remove from each obstacle type. Check `world_model_base.objects` for the correct number of obstacles.

will-44, 2024-11-22:
Thank you for your answer !
Currently, when I add a cuboid as an obstacle, even if I remove it and update the world config cache, the collision still appears in voxels and the path planner still see it. Do you have any ideas ?
An example of the code:
```
world_cfg.remove_obstacle(name)
motion_gen.update_world(world_cfg)
motion_gen.world_coll_checker.clear_cache() 
```

closed 2025-04-25 by balakumar-s