NVlabs/curobo #385: isaac sim crashed when run "omni_python examples/isaac_sim/motion_gen_reacher.py to run a motion generation example"

shaohedu, 2024-09-12:
If it’s not a bug, please use discussions: https://github.com/NVlabs/curobo/discussions

Please provide the below information in addition to your issue:

1. cuRobo installation mode (isaac sim):
2. python version: Python 3.10.14
3. Isaac Sim version (if using): 4.0.0

------
Issue Details
I follwed the introduction of the link (https://curobo.org/get_started/1_install_instructions.html#install-for-use-in-isaac-sim) to build and run curobo, I can build curobo succesfully, but when run the example of "omni_python examples/isaac_sim/motion_gen_reacher.py to run a motion generation example", it blocked and got the follwing print:
[image]
isaac sim also stuck in the follwing interface:
[image]

anyone can help? thanks very much~

AmirpooyaSh, 2024-10-09:
@shaohedu 
That's basically the same issue I faced earlier.
It's all the matter of time. You should wait until it comes up (Takes like 5-6 mins to launch Isaac-Sim using a single RTX 2070 Super).
It should come up unless you receive python-based errors which should be related to the "motion_gen_reacher.py".

I would appreciate it if the committers suggest faster bring up solutions though.

Xin-jue, 2024-12-27:
@AmirpooyaSh 
[image]
I also encountered this problem where the Isaac window opened but nothing was loaded. But my situation is that the window will close after waiting for a few seconds. Would you happen to know how to solve this?

AmirpooyaSh, 2025-01-03:
@Xin-jue 

Could you provide the whole log from the terminal? These are just some "carb library" warnings that should not cause the problem. I suspect that something else is causing the closure (further down in the terminal logs), so just send the whole log and I'll take a look at it ...

Xin-jue, 2025-01-15:
@AmirpooyaSh 

Thanks！The complete terminal operation report is as follows：

Warning: running in conda env, please deactivate before executing this script
If conda is desired please source setup_conda_env.sh in your python 3.10 conda env and run python normally
Starting kit application with the following args:  ['/home/yons/omniverse/isaac-sim-4.0.0/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py', '/home/yons/omniverse/isaac-sim-4.0.0/apps/omni.isaac.sim.python.kit', '--/app/tokens/exe-path=/home/yons/omniverse/isaac-sim-4.0.0/kit', '--/persistent/app/viewport/displayOptions=3094', '--/rtx/materialDb/syncLoads=True', '--/rtx/hydra/materialSyncLoads=True', '--/omni.kit.plugin/syncUsdLoads=True', '--/app/renderer/resolution/width=1920', '--/app/renderer/resolution/height=1080', '--/app/window/width=1440', '--/app/window/height=900', '--/renderer/multiGpu/enabled=True', '--/app/fastShutdown=True', '--ext-folder', '/home/yons/omniverse/isaac-sim-4.0.0/exts', '--ext-folder', '/home/yons/omniverse/isaac-sim-4.0.0/apps', '--/physics/cudaDevice=0', '--portable']
Passing the following args to the base kit application:  []
[Info] [carb] Logging to file: /home/yons/omniverse/isaac-sim-4.0.0/kit/logs/Kit/Isaac-Sim/4.0/kit_20250115_192028.log
2025-01-15 11:20:28 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
[0.035s] [ext: omni.kit.async_engine-0.0.0] startup
[0.210s] [ext: omni.stats-0.0.0] startup
[0.211s] [ext: omni.client-1.1.0] startup
[0.225s] [ext: omni.datastore-0.0.0] startup
[0.226s] [ext: omni.blobkey-1.1.0] startup
[0.226s] [ext: omni.ujitso.default-1.0.0] startup
[0.226s] [ext: omni.hsscclient-0.0.0] startup
[0.226s] [ext: omni.rtx.shadercache.vulkan-1.0.0] startup
[0.227s] [ext: omni.assets.plugins-0.0.0] startup
[0.228s] [ext: omni.gpu_foundation-0.0.0] startup
[0.232s] [ext: carb.windowing.plugins-1.0.0] startup
[0.236s] [ext: omni.kit.renderer.init-0.0.0] startup

|---------------------------------------------------------------------------------------------|
| Driver Version: 535.183.01    | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 4090          | Yes: 0 |     | 24810   MB | 10de      | 0          |
|     |                                  |        |     |            | 2684      | 38a5b9bb.. |
|     |                                  |        |     |            | 1         |            |
|---------------------------------------------------------------------------------------------|
| 1   | NVIDIA GeForce RTX 4090          | Yes: 1 |     | 24810   MB | 10de      | 0          |
|     |                                  |        |     |            | 2684      | 01057dde.. |
|     |                                  |        |     |            | 6         |            |
|=============================================================================================|
| OS: 20.04.6 LTS (Focal Fossa) ubuntu, Version: 20.04.6, Kernel: 5.15.0-119-generic
| XServer Vendor: The X.Org Foundation, XServer Version: 12013000 (1.20.13.0)
| Processor: Intel(R) Core(TM) i9-14900KF | Cores: 24 | Logical: 32
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 128636 | Free Memory: 108795
| Total Page/Swap (MB): 2047 | Free Page/Swap: 2047
|---------------------------------------------------------------------------------------------|
[1.055s] [ext: omni.kit.pipapi-0.0.0] startup
[1.056s] [ext: omni.kit.pip_archive-0.0.0] startup
[1.056s] [ext: omni.pip.compute-1.3.1] startup
[1.056s] [ext: omni.pip.cloud-1.1.3] startup
[1.058s] [ext: omni.isaac.core_archive-2.3.0] startup
[1.058s] [ext: omni.isaac.ml_archive-2.0.1] startup
[1.058s] [ext: omni.mtlx-0.1.0] startup
[1.058s] [ext: omni.usd.config-1.0.3] startup
[1.059s] [ext: omni.gpucompute.plugins-0.0.0] startup
[1.060s] [ext: omni.usd.libs-1.0.0] startup
[1.142s] [ext: omni.kit.telemetry-0.5.0] startup
[1.165s] [ext: omni.kit.loop-isaac-1.2.0] startup
[1.166s] [ext: omni.kit.test-0.0.0] startup
[1.192s] [ext: omni.appwindow-1.1.8] startup
[1.629s] [ext: omni.kit.renderer.core-1.0.1] startup
[3.796s] [ext: omni.kit.renderer.capture-0.0.0] startup
[3.797s] [ext: omni.kit.renderer.imgui-1.0.1] startup
[3.868s] [ext: omni.ui-2.23.11] startup
[3.875s] [ext: omni.kit.mainwindow-1.0.3] startup
[3.875s] [ext: carb.audio-0.1.0] startup
[3.897s] [ext: omni.uiaudio-1.0.0] startup
[3.897s] [ext: omni.kit.uiapp-0.0.0] startup
[3.897s] [ext: omni.usd.schema.forcefield-106.0.16] startup
[3.985s] [ext: omni.usd.schema.scene.visualization-2.0.2] startup
[3.986s] [ext: omni.usd.schema.audio-0.0.0] startup
[... 282 lines cut ...]
    self.sample_lib = SampleLib(self.sample_params)
  File "/home/yons/Isaac/isaac_sim_motion_generator/curobo/src/curobo/util/sample_lib.py", line 350, in __init__
    self.random_sample_lib = RandomSampleLib(sample_config)
  File "/home/yons/Isaac/isaac_sim_motion_generator/curobo/src/curobo/util/sample_lib.py", line 218, in __init__
[... cut ...]
2025-01-15 11:20:40 [11,605ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-01-15 11:20:40 [11,605ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-01-15 11:20:40 [11,605ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-01-15 11:20:40 [11,605ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Core' for removal
2025-01-15 11:20:40 [11,809ms] [Warning] [carb] Recursive unloadAllPlugins() detected!

AmirpooyaSh, 2025-01-24:
@Xin-jue 

It seems that "guard_size_oblivious" can't be found within the torch library. I suspect that the PyTorch installed within the Isaac sim virtual environment is outdated or mismatches the version required to run CuRobo.

Also, I've seen it earlier in the logs that your CUDA version is 11.5. I'm pretty sure that CuRobo needs 11.8 (which made us to migrate from Ubuntu 20 to 22 and start building our ROS workspace within a Noetic docker).

I suggest you to change your CUDA version to 11.8. If it didn't work out, try to remove pytorch from the Isaac sim virtual environment and relaunch the curobo installation script (cd curobo/ && omni_python -m pip install -e .[isaacsim] --no-build-isolation).

Let me know if it fixed the issue ^-^.

aditivSC, 2025-02-19:
Hello, 

I am also having similar issues. I followed the instructions for installing curobo for use in isaac sim according to this [link.](https://curobo.org/get_started/1_install_instructions.html#install-for-use-in-isaac-sim)

I have installed the cuda 11.8 toolkit. I get the following output log when I try to run the motion_gen_reacher.py example:

`[Warning] [omni.isaac.kit] omni.isaac.kit has been deprecated in favor of isaacsim.simulation_app. Please update your code accordingly.
Starting kit application with the following args:  ['/home/tarantula/isaacsim/exts/isaacsim.simulation_app/isaacsim/simulation_app/simulation_app.py', '/home/tarantula/isaacsim/apps/isaacsim.exp.base.python.kit', '--/app/tokens/exe-path=/home/tarantula/isaacsim/kit', '--/persistent/app/viewport/displayOptions=3094', '--/rtx/materialDb/syncLoads=True', '--/rtx/hydra/materialSyncLoads=True', '--/omni.kit.plugin/syncUsdLoads=True', '--/app/renderer/resolution/width=1920', '--/app/renderer/resolution/height=1080', '--/app/window/width=1440', '--/app/window/height=900', '--/renderer/multiGpu/enabled=True', '--/app/fastShutdown=True', '--/app/installSignalHandlers=0', '--ext-folder', '/home/tarantula/isaacsim/exts', '--ext-folder', '/home/tarantula/isaacsim/apps', '--/physics/cudaDevice=0', '--portable']
Passing the following args to the base kit application:  []
[Info] [carb] Logging to file: /home/tarantula/isaacsim/kit/logs/Kit/Isaac-Sim Python/4.5/kit_20250219_141139.log
2025-02-19 13:11:39 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
[0.047s] [ext: omni.kit.async_engine-0.0.1] startup
[0.486s] [ext: omni.metrics.core-0.0.1] startup
[0.486s] [ext: omni.client.lib-1.0.0] startup
[0.506s] [ext: omni.blobkey-1.1.2] startup
[0.506s] [ext: omni.stats-1.0.1] startup
[0.507s] [ext: omni.datastore-0.0.0] startup
[0.512s] [ext: omni.client-1.2.2] startup
[0.516s] [ext: omni.ujitso.default-1.0.0] startup
[0.518s] [ext: omni.hsscclient-1.1.1] startup
2025-02-19 13:11:39 [504ms] [Warning] [omni.datastore] OmniHub is inaccessible
[0.520s] [ext: omni.gpu_foundation.shadercache.vulkan-1.0.0] startup
[0.522s] [ext: omni.assets.plugins-0.0.0] startup
[0.523s] [ext: omni.gpu_foundation-0.0.0] startup
[0.530s] [ext: carb.windowing.plugins-1.0.0] startup
[0.545s] [ext: omni.kit.renderer.init-0.0.0] startup

|---------------------------------------------------------------------------------------------|
| Driver Version: 550.120       | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 4090          | Yes: 0 |     | 24810   MB | 10de      | 0          |
|     |                                  |        |     |            | 2684      | ac4488fb.. |
|     |                                  |        |     |            | 9         |            |
|=============================================================================================|
| OS: 22.04.5 LTS (Jammy Jellyfish) ubuntu, Version: 22.04.5, Kernel: 6.8.0-52-generic
| XServer Vendor: The X.Org Foundation, XServer Version: 12101004 (1.21.1.4)
| Processor: AMD Ryzen 9 5950X 16-Core Processor
| Cores: 16 | Logical Cores: 32
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 31996 | Free Memory: 24273
| Total Page/Swap (MB): 10504 | Free Page/Swap: 7756
|---------------------------------------------------------------------------------------------|
2025-02-19 13:11:40 [870ms] [Warning] [gpu.foundation.plugin] IOMMU is enabled.
[1.048s] [ext: omni.kit.pipapi-0.0.0] startup
[1.049s] [ext: omni.kit.pip_archive-0.0.0] startup
[1.050s] [ext: omni.pip.compute-1.4.5] startup
[1.050s] [ext: omni.materialx.libs-1.0.6] startup
[1.058s] [ext: omni.kit.telemetry-0.5.1] startup
[1.084s] [ext: omni.pip.cloud-1.1.7] startup
[1.087s] [ext: omni.isaac.core_archive-2.3.3] startup
[1.088s] [ext: omni.usd.config-1.0.5] startup
[1.093s] [ext: omni.gpucompute.plugins-0.0.0] startup
[1.094s] [ext: omni.usd.libs-1.0.1] startup
[1.171s] [ext: omni.isaac.ml_archive-2.1.2] startup
[1.172s] [ext: omni.mdl-55.0.1] startup
[1.190s] [ext: omni.iray.libs-0.0.0] startup
[1.195s] [ext: omni.mdl.neuraylib-0.2.10] startup
[1.196s] [ext: omni.kit.usd.mdl-1.0.7] startup
[1.285s] [ext: omni.kit.loop-isaac-1.2.1] startup
[1.286s] [ext: omni.kit.test-1.1.2] startup
[1.322s] [ext: omni.appwindow-1.1.9] startup
[1.665s] [ext: omni.kit.renderer.core-1.0.2] startup
[2.340s] [ext: omni.kit.renderer.capture-0.0.0] startup
[2.343s] [ext: omni.kit.renderer.imgui-1.0.2] startup
[2.403s] [ext: omni.ui-2.26.5] startup
[2.413s] [ext: omni.kit.mainwindow-1.0.3] startup
[... 370 lines cut ...]
2025-02-19 13:11:47 [7,782ms] [Error] [omni.kit.app._impl] [py stderr]:   File "/home/tarantula/workspace/curobo/examples/isaac_sim/helper.py", line 19, in <module>
2025-02-19 13:11:47 [7,782ms] [Error] [omni.kit.app._impl] [py stderr]:     from omni.isaac.core.materials import OmniPBR
[... cut ...]
2025-02-19 13:11:47 [7,803ms] [Warning] [omni.fabric.plugin] gFabricState->gUsdStageToSimStageWithHistoryMap had 1 outstanding SimStageWithHistory(s) at shutdown
2025-02-19 13:11:47 [7,803ms] [Warning] [carb.audio.context] 1 contexts were leaked
2025-02-19 13:11:47 [7,921ms] [Warning] [carb] Recursive unloadAllPlugins() detected!

`
It would be great if you could help me out. I am getting confused with the document versioning. The examples are linked to Isaac sim 4.20. I have Isaac sim 4.50 installed, is that why none of the curobo examples are working for me?

AmirpooyaSh, 2025-02-19:
@aditivSC,

I just saw the new release :D.

CuRobo examples are going to need some modifications for sure (the least is the library annotation updates).

If you want to proceed sooner than their release (if so), checkout this link (Migration Guide) and do the changes within the code accordingly. (https://github.com/isaac-sim/IsaacLab/releases)

aditivSC, 2025-02-19:
@AmirpooyaSh,

Thanks for the heads up about the new release. Unfortunately, I think it is too much work for me to make these changes to my code at this stage. I am going to try install version 4.0.0 of Isaac sim as recommended in the curobo documentation and see if that helps.

AmirpooyaSh, 2025-02-19:
@aditivSC 

I'm not sure about Isaac Sim 4.0.0. I started working with CuRobo in September 2024 and I kept using 4.2.0 so far without any major issues. The only modification I had to do within their installation instructions was modifying the "omni_python" venv path in bashrc:

Their Original Command:
echo "alias omni_python='~/.local/share/ov/pkg/isaac_sim-4.0.0/python.sh'" >> ~/.bashrc

I just changed the Isaac Sim version:
echo "alias omni_python='~/.local/share/ov/pkg/isaac-sim-4.2.0/python.sh'">> ~/.bashrc

I belive 4.0.0 would solve your issues, but if it didn't go for 4.2.0. This version is what I'm confident in as of now.

balakumar-s, 2025-04-25:
We have now updated the cuRobo examples to work with Isaac Sim 4.5. Please re-open if issue still persists.

closed 2025-04-25 by balakumar-s