NVlabs/curobo #397: Collision Object not detected

Sajjan2406, 2024-10-01:
If it’s not a bug, please use discussions: https://github.com/NVlabs/curobo/discussions

Please provide the below information in addition to your issue:

1. cuRobo installation mode (choose from [python, isaac sim, docker python, docker isaac sim]): Isaac ROS Cumotion(docker)
2. python version:
3. Isaac Sim version (if using): 2023.1.1

------
Issue Details
i am spawning the robot and the scenario through the .xacro file while launching 'move_group' node. I am able to visualize the robot and scene in the Rviz window. 
1. When i start Planning using 'OMPL', the planner detects the scene(in this case bookshelf) as an obstacle and plans accordingly
2. When i use 'cumotion' to do the planning, the planning happens without considering the bookshelf. When the plan is sent back to 'move_group', there is an error: collision detected between the robot and the bookshelf

this is how i run my cumotion planner node from CLI: 'ros2 run isaac_ros_cumotion cumotion_planner_node --ros-args -p robot:=franka.xrdf -p urdf_path:=/opt/ros/humble/share/moveit_resources_panda_description/urdf/panda.urdf'

I thought the problem was arising because i did not update the urdf file that includes the bookshelf too, during the launch of 'cumotion planner node'. So i updated the urdf file by combining both the urdf using a CLI. Even now with the updated urdf, the error is the same

Please let me know how to proceed with this error.

this is the error logged in the Move_group terminal:
[move_group-5] [INFO] [1727707475.384943894] [moveit_collision_detection_fcl.collision_common]: Found a contact between 'panda_hand' (type 'Robot link') and 'bookshelf_top' (type 'Robot link'), which constitutes a collision. Contact information is not stored. 
[move_group-5] [INFO] [1727707475.384966718] [moveit_collision_detection_fcl.collision_common]: Collision checking is considered complete (collision was found and 0 contacts are stored) 
[move_group-5] [INFO] [1727707475.385036370] [moveit_collision_detection_fcl.collision_common]: Found a contact between 'panda_hand' (type 'Robot link') and 'bookshelf_top' (type 'Robot link'), which constitutes a collision. Contact information is not stored. 
[move_group-5] [INFO] [1727707475.385041325] [moveit_collision_detection_fcl.collision_common]: Collision checking is considered complete (collision was found and 0 contacts are stored) 
[move_group-5] [INFO] [1727707475.385095727] [moveit_collision_detection_fcl.collision_common]: Found a contact between 'panda_hand' (type 'Robot link') and 'bookshelf_top' (type 'Robot link'), which constitutes a collision. Contact information is not stored. 
[move_group-5] [INFO] [1727707475.385099960] [moveit_collision_detection_fcl.collision_common]: Collision checking is considered complete (collision was found and 0 contacts are stored) 
[move_group-5] [INFO] [1727707475.385177073] [moveit_collision_detection_fcl.collision_common]: Found a contact between 'panda_hand' (type 'Robot link') and 'bookshelf_top' (type 'Robot link'), which constitutes a collision. Contact information is not stored. 
[move_group-5] [INFO] [1727707475.385186044] [moveit_collision_detection_fcl.collision_common]: Collision checking is considered complete (collision was found and 0 contacts are stored) 
[move_group-5] [INFO] [1727707475.385230716] [moveit_collision_detection_fcl.collision_common]: Found a contact between 'panda_hand' (type 'Robot link') and 'bookshelf_top' (type 'Robot link'), which constitutes a collision. Contact information is not stored. 
[move_group-5] [INFO] [1727707475.385232859] [moveit_collision_detection_fcl.collision_common]: Collision checking is considered complete (collision was found and 0 contacts are stored) 
[move_group-5] [INFO] [1727707475.385282138] [moveit_collision_detection_fcl.collision_common]: Found a contact between 'bookshelf_top' (type 'Robot link') and 'panda_hand' (type 'Robot link'), which constitutes a collision. Contact information is not stored. 
[move_group-5] [INFO] [1727707475.385283844] [moveit_collision_detection_fcl.collision_common]: Collision checking is considered complete (collision was found and 0 contacts are stored) 
[move_group-5] [INFO] [1727707475.385311112] [moveit_collision_detection_fcl.collision_common]: Found a contact between 'bookshelf_top' (type 'Robot link') and 'panda_hand' (type 'Robot link'), which constitutes a collision. Contact information is not stored. 
[move_group-5] [INFO] [1727707475.385312736] [moveit_collision_detection_fcl.collision_common]: Collision checking is considered complete (collision was found and 0 contacts are stored) 
[move_group-5] [INFO] [1727707475.385339259] [moveit_collision_detection_fcl.collision_common]: Found a contact between 'bookshelf_top' (type 'Robot link') and 'panda_hand' (type 'Robot link'), which constitutes a collision. Contact information is not stored. 
[move_group-5] [INFO] [1727707475.385340871] [moveit_collision_detection_fcl.collision_common]: Collision checking is considered complete (collision was found and 0 contacts are stored) 
[move_group-5] [INFO] [1727707475.385364934] [moveit_collision_detection_fcl.collision_common]: Found a contact between 'bookshelf_top' (type 'Robot link') and 'panda_hand' (type 'Robot link'), which constitutes a collision. Contact information is not stored. 
[move_group-5] [INFO] [1727707475.385366306] [moveit_collision_detection_fcl.collision_common]: Collision checking is considered complete (collision was found and 0 contacts are stored) 
[move_group-5] [INFO] [1727707475.385401921] [moveit_collision_detection_fcl.collision_common]: Found a contact between 'bookshelf_top' (type 'Robot link') and 'panda_hand' (type 'Robot link'), which constitutes a collision. Contact information is not stored. 
[move_group-5] [INFO] [1727707475.385403372] [moveit_collision_detection_fcl.collision_common]: Collision checking is considered complete (collision was found and 0 contacts are stored) 
[move_group-5] [INFO] [1727707475.385427605] [moveit_collision_detection_fcl.collision_common]: Found a contact between 'bookshelf_top' (type 'Robot link') and 'panda_hand' (type 'Robot link'), which constitutes a collision. Contact information is not stored. 
[move_group-5] [INFO] [1727707475.385437186] [moveit_collision_detection_fcl.collision_common]: Collision checking is considered complete (collision was found and 0 contacts are stored) 
[move_group-5] [INFO] [1727707475.385460718] [moveit_collision_detection_fcl.collision_common]: Found a contact between 'bookshelf_top' (type 'Robot link') and 'panda_hand' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[move_group-5] [INFO] [1727707475.385462080] [moveit_collision_detection_fcl.collision_common]: Collision checking is considered complete (collision was found and 0 contacts are stored) 
[move_group-5] [INFO] [1727707475.385485721] [moveit_collision_detection_fcl.collision_common]: Found a contact between 'bookshelf_top' (type 'Robot link') and 'panda_hand' (type 'Robot link'), which constitutes a collision. Contact information is not stored. 
[move_group-5] [INFO] [1727707475.385487039] [moveit_collision_detection_fcl.collision_common]: Collision checking is considered complete (collision was found and 0 contacts are stored) 
[move_group-5] [ERROR] [1727707475.385503778] [moveit.ros_planning.planning_pipeline]: Completed listing of explanations for invalid states. 
[move_group-5] [INFO] [1727707475.385610928] [moveit_move_group_default_capabilities.move_action_capability]: Motion plan was found but it seems to be invalid (possibly due to postprocessing). Not executing. 
[rviz2-1] [INFO] [1727707475.385789082] [move_group_interface]: Planning request aborted 
[rviz2-1] [ERROR] [1727707475.386361721] [move_group_interface]: MoveGroupInterface::plan() failed or timeout reached

Alice-ly, 2024-12-31:
Hi, I have also met similar problems when curobo generate trajectory without considering obstacles but move group can detect the collision, and I have tried the parameters <https://github.com/NVlabs/curobo/issues/450>, some did work but have not solved the problem totally.

balakumar-s, 2025-01-04:
Reading obstacles from urdf is not supported. Use this workaround instead: https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_cumotion/issues/20#issuecomment-2407239939

closed 2025-01-04 by balakumar-s