NVlabs/curobo #398: Docker Build issue on Jetson Orin AGX

yashwant-ies, 2024-10-03:
Unable to setup curobo on Jetson orin AGX, while building with Docker. I see below error:

```

nvblox_torch# python3 -m pip install -e .
Obtaining file:///pkgs/nvblox_torch
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python3 /tmp/tmpa7amxe9n get_requires_for_build_wheel /tmp/tmpxwxtkr88
       cwd: /pkgs/nvblox_torch
  Complete output (81 lines):
  running egg_info
  creating src/nvblox_torch.egg-info
  writing src/nvblox_torch.egg-info/PKG-INFO
  writing dependency_links to src/nvblox_torch.egg-info/dependency_links.txt
  writing requirements to src/nvblox_torch.egg-info/requires.txt
  writing top-level names to src/nvblox_torch.egg-info/top_level.txt
  writing manifest file 'src/nvblox_torch.egg-info/SOURCES.txt'
  reading manifest template 'MANIFEST.in'
  /tmp/pip-build-env-a8op_3tb/overlay/lib/python3.8/site-packages/setuptools/dist.py:452: SetuptoolsDeprecationWarning: Invalid dash-separated options
  !!
  
          ********************************************************************************
          Usage of dash-separated 'long-description-content-type' will not be supported in future
          versions. Please use the underscore name 'long_description_content_type' instead.
  
          This deprecation is overdue, please update your project and remove deprecated
          calls to avoid build errors in the future.
  
          See https://setuptools.pypa.io/en/latest/userguide/declarative_config.html for details.
          ********************************************************************************
  
  !!
    opt = self.warn_dash_deprecation(opt, section)
  /tmp/pip-build-env-a8op_3tb/overlay/lib/python3.8/site-packages/setuptools/dist.py:452: SetuptoolsDeprecationWarning: Invalid dash-separated options
  !!
  
          ********************************************************************************
          Usage of dash-separated 'long-description-content-type' will not be supported in future
          versions. Please use the underscore name 'long_description_content_type' instead.
  
          This deprecation is overdue, please update your project and remove deprecated
          calls to avoid build errors in the future.
  
          See https://setuptools.pypa.io/en/latest/userguide/declarative_config.html for details.
          ********************************************************************************

  
  !!
    opt = self.warn_dash_deprecation(opt, section)
  warning: no directories found matching 'src/nvblox_torch/bin/*.so'
  adding license file 'LICENSE'
  Traceback (most recent call last):
    File "/tmp/tmpa7amxe9n", line 280, in <module>
      main()
    File "/tmp/tmpa7amxe9n", line 263, in main
      json_out['return_val'] = hook(**hook_input['kwargs'])
    File "/tmp/tmpa7amxe9n", line 114, in get_requires_for_build_wheel
      return hook(config_settings)
    File "/tmp/pip-build-env-a8op_3tb/overlay/lib/python3.8/site-packages/setuptools/build_meta.py", line 332, in get_requires_for_build_wheel
      return self._get_build_requires(config_settings, requirements=[])
    File "/tmp/pip-build-env-a8op_3tb/overlay/lib/python3.8/site-packages/setuptools/build_meta.py", line 302, in _get_build_requires
      self.run_setup()
    File "/tmp/pip-build-env-a8op_3tb/overlay/lib/python3.8/site-packages/setuptools/build_meta.py", line 318, in run_setup
      exec(code, locals())
    File "<string>", line 13, in <module>
    File "/tmp/pip-build-env-a8op_3tb/overlay/lib/python3.8/site-packages/setuptools/__init__.py", line 117, in setup
      return distutils.core.setup(**attrs)
    File "/tmp/pip-build-env-a8op_3tb/overlay/lib/python3.8/site-packages/setuptools/_distutils/core.py", line 183, in setup
      return run_commands(dist)
    File "/tmp/pip-build-env-a8op_3tb/overlay/lib/python3.8/site-packages/setuptools/_distutils/core.py", line 199, in run_commands
      dist.run_commands()
    File "/tmp/pip-build-env-a8op_3tb/overlay/lib/python3.8/site-packages/setuptools/_distutils/dist.py", line 954, in run_commands
      self.run_command(cmd)
    File "/tmp/pip-build-env-a8op_3tb/overlay/lib/python3.8/site-packages/setuptools/dist.py", line 950, in run_command
      super().run_command(command)
    File "/tmp/pip-build-env-a8op_3tb/overlay/lib/python3.8/site-packages/setuptools/_distutils/dist.py", line 973, in run_command
      cmd_obj.run()
    File "/tmp/pip-build-env-a8op_3tb/overlay/lib/python3.8/site-packages/setuptools/command/egg_info.py", line 311, in run
      self.find_sources()
    File "/tmp/pip-build-env-a8op_3tb/overlay/lib/python3.8/site-packages/setuptools/command/egg_info.py", line 319, in find_sources
      mm.run()
    File "/tmp/pip-build-env-a8op_3tb/overlay/lib/python3.8/site-packages/setuptools/command/egg_info.py", line 545, in run
      self.prune_file_list()
    File "/tmp/pip-build-env-a8op_3tb/overlay/lib/python3.8/site-packages/setuptools/command/sdist.py", line 161, in prune_file_list
      super().prune_file_list()
    File "/tmp/pip-build-env-a8op_3tb/overlay/lib/python3.8/site-packages/setuptools/_distutils/command/sdist.py", line 380, in prune_file_list
      base_dir = self.distribution.get_fullname()
    File "/tmp/pip-build-env-a8op_3tb/overlay/lib/python3.8/site-packages/setuptools/_core_metadata.py", line 267, in get_fullname
      return _distribution_fullname(self.get_name(), self.get_version())
    File "/tmp/pip-build-env-a8op_3tb/overlay/lib/python3.8/site-packages/setuptools/_core_metadata.py", line 285, in _distribution_fullname
      canonicalize_version(version, strip_trailing_zero=False),
  TypeError: canonicalize_version() got an unexpected keyword argument 'strip_trailing_zero'
  ----------------------------------------
ERROR: Command errored out with exit status 1: /usr/bin/python3 /tmp/tmpa7amxe9n get_requires_for_build_wheel /tmp/tmpxwxtkr88 Check the logs for full command output.

```

yashwant-ies, 2024-10-03:
In case I proceeded docker without this nvblox_torch and trying the kinematics scripts, it is giving me this error!

```
python3 forward_kinematics.py 
Traceback (most recent call last):
  File "forward_kinematics.py", line 25, in <module>
    robot_cfg = RobotConfig.from_basic(urdf_file, base_link, ee_link, tensor_args)
  File "/usr/local/lib/python3.8/dist-packages/curobo/types/robot.py", line 58, in from_basic
    cuda_robot_model_config = CudaRobotModelConfig.from_basic_urdf(
  File "/usr/local/lib/python3.8/dist-packages/curobo/cuda_robot_model/cuda_robot_model.py", line 125, in from_basic_urdf
    return CudaRobotModelConfig.from_config(config)
  File "/usr/local/lib/python3.8/dist-packages/curobo/cuda_robot_model/cuda_robot_model.py", line 248, in from_config
    generator = CudaRobotGenerator(config)
  File "/usr/local/lib/python3.8/dist-packages/curobo/cuda_robot_model/cuda_robot_generator.py", line 260, in __init__
    self.initialize_tensors()
  File "/usr/lib/python3.8/contextlib.py", line 75, in inner
    return func(*args, **kwds)
  File "/usr/local/lib/python3.8/dist-packages/curobo/cuda_robot_model/cuda_robot_generator.py", line 285, in initialize_tensors
    self._link_spheres_tensor = torch.empty(
  File "/usr/local/lib/python3.8/dist-packages/torch/cuda/__init__.py", line 217, in _lazy_init
    torch._C._cuda_init()
RuntimeError: Found no NVIDIA driver on your system. Please check that you have an NVIDIA GPU and installed a driver from http://www.nvidia.com/Download/index.aspx
```

benoit-robotics, 2024-10-07:
Adding this line to aarch64.dockerfile seems to fix the issue:

> RUN python3 -m pip install packaging==22

sarthak-ies, 2024-10-08:
Hi @benoit-robotics after this installation, the first error is solved but the second error still persists.
Using this [script](https://curobo.org/get_started/2a_python_examples.html#forward-kinematics) to run inside docker, producing this error
```
root@ros1-docker:/pkgs# python3 forward_kinematics.py 
Traceback (most recent call last):
  File "forward_kinematics.py", line 25, in <module>
    robot_cfg = RobotConfig.from_basic(urdf_file, base_link, ee_link, tensor_args)
  File "/usr/local/lib/python3.8/dist-packages/curobo/types/robot.py", line 58, in from_basic
    cuda_robot_model_config = CudaRobotModelConfig.from_basic_urdf(
  File "/usr/local/lib/python3.8/dist-packages/curobo/cuda_robot_model/cuda_robot_model.py", line 125, in from_basic_urdf
    return CudaRobotModelConfig.from_config(config)
  File "/usr/local/lib/python3.8/dist-packages/curobo/cuda_robot_model/cuda_robot_model.py", line 248, in from_config
    generator = CudaRobotGenerator(config)
  File "/usr/local/lib/python3.8/dist-packages/curobo/cuda_robot_model/cuda_robot_generator.py", line 260, in __init__
    self.initialize_tensors()
  File "/usr/lib/python3.8/contextlib.py", line 75, in inner
    return func(*args, **kwds)
  File "/usr/local/lib/python3.8/dist-packages/curobo/cuda_robot_model/cuda_robot_generator.py", line 285, in initialize_tensors
    self._link_spheres_tensor = torch.empty(
  File "/usr/local/lib/python3.8/dist-packages/torch/cuda/__init__.py", line 217, in _lazy_init
    torch._C._cuda_init()
RuntimeError: Found no NVIDIA driver on your system. Please check that you have an NVIDIA GPU and installed a driver from http://www.nvidia.com/Download/index.aspx
```

balakumar-s, 2025-04-25:
Would it be possible to start from either a pytorch docker or follow these steps to setup the isaac ros container on jetson:
https://nvidia-isaac-ros.github.io/getting_started/hardware_setup/compute/index.html#jetson-platforms

jarkenau, 2025-08-20:
I’m experiencing the same problem where the driver can’t be found when using the [aarch64 Dockerfile](https://github.com/NVlabs/curobo/blob/main/docker/aarch64.dockerfile) provided by cuRubo. I built the image with:

```bash
bash build_docker.sh aarch64
```

and started it using:

```bash
bash start_docker_aarch64.sh
```

I’m running JetPack 6.2.1 and have followed all the instructions mentioned above.

For reference, the Docker [isaac_ros_dev environment](https://nvidia-isaac-ros.github.io/concepts/docker_devenv/index.html#development-environment) works perfectly on my system, so it seems like the issue is specific to the Docker setup here.

closed 2026-05-18 by balakumar-s