NVlabs/curobo #404: UsdKinematicsParser flip_joint_limits wrong behavior

ChengshuLi, 2024-10-08:
1. cuRobo installation mode: python
2. python version: 3.10
3. Isaac Sim version (if using): 4.1.0 

------
When testing with Franka Panda, it noticed that `UsdKinematicsParser` `flip_joint_limits` has the wrong behavior. Specifically, instead of 
```
if joint_name in self._flip_joint_limits:
    joint_limits = np.ravel(
        [-1.0 * j_prim.GetUpperLimitAttr().Get(), j_prim.GetLowerLimitAttr().Get()]
    )
```
it should be 
```
if joint_name in self._flip_joint_limits:
    joint_offset = [-1.0, 0.0]
```

I found out about this when looking at the `panda_finger_joint2` link, with the following visualization.
```
sph = mg.kinematics.get_robot_as_spheres(joint_pos)
WorldConfig(sphere=sph[0]).save_world_as_mesh("/scr/chengshu/Downloads/robot.obj")
```

Before:
[image]

After:
[image]

Is my fix reasonable? Or maybe I wasn't passing in the argument correctly?

Many thanks!

balakumar-s, 2024-10-08:
We recommend using the URDF kinematics parser. The USD parser is not fully implemented and does not work on most robot USD files.

ChengshuLi, 2024-10-08:
@balakumar-s Thanks for the quick reply!

I understand URDF kinematics parser is more mature. However, currently I have some custom changes applied to the USD file that is not found in URDF (i.e. I manually changed stuff on the USD in Isaac Sim and saved it). Therefore, I might need to use the USD kinematics parser for the most-up-to-date information.

balakumar-s, 2025-04-25:
Is it possible to convert the usd to urdf before sending to cuRobo? The USDKinematicParser is not being actively worked on. There are several gaps in the current implementation.

closed 2025-04-25 by balakumar-s