NVlabs/curobo #413: VULKAN_SDK_VERSION  in isaac_sim.dockerfile expired

pmj110119, 2024-10-15:
If it’s not a bug, please use discussions: https://github.com/NVlabs/curobo/discussions

Please provide the below information in addition to your issue:

1. cuRobo installation mode (choose from [python, isaac sim, docker python, docker isaac sim]):
2. python version:
3. Isaac Sim version (if using): 

------
Issue Details

isaac_sim.dockerfile will build with error because:
```
ARG VULKAN_SDK_VERSION=1.3.224.1
```

which has expired in  `hhttps://sdk.lunarg.com/sdk/download/1.3.224.1/linux/vulkansdk-linux-x86_64-1.3.224.1.tar.gz`. 

We can edit dockerfile:
```
ARG VULKAN_SDK_VERSION=1.3.231.1
```

sesem738, 2025-03-27:
Is it possible to update the files with this solution? It's still ARG VULKAN_SDK_VERSION=1.3.224.1 in the docs

PostResPerditas, 2025-11-20:
record the problems and corresponding solutions I meet.
ERROR: failed to solve: process "/bin/sh -c wget -q --show-progress     --progress=bar:force:noscroll     https://sdk.lunarg.com/sdk/download/${VULKAN_SDK_VERSION}/linux/vulkansdk-linux-x86_64-${VULKAN_SDK_VERSION}.tar.gz
and
cp: cannot stat '/opt/vulkan/1.3.280.1/x86_64/etc/vulkan/explicit_layer.d/VkLayer_*.json': No such file or directory

it seems that the version of the vulkan sdk has been changed, also did the path fo VkLayer_*.json in the new version of vulkan sdk.

so I change the code Line 27 and Line 102 in docker/isaac_sim.dockerfile, that:

ARG VULKAN_SDK_VERSION=1.3.280.1

RUN wget -q --show-progress \
    --progress=bar:force:noscroll \
    https://sdk.lunarg.com/sdk/download/${VULKAN_SDK_VERSION}/linux/vulkansdk-linux-x86_64-${VULKAN_SDK_VERSION}.tar.xz \
    -O /tmp/vulkansdk-linux-x86_64-${VULKAN_SDK_VERSION}.tar.xz \
    && echo "Installing Vulkan SDK ${VULKAN_SDK_VERSION}" \
    && mkdir -p /opt/vulkan \
    && tar -xJvf /tmp/vulkansdk-linux-x86_64-${VULKAN_SDK_VERSION}.tar.xz -C /opt/vulkan \
    && mkdir -p /usr/local/include/ && cp -ra /opt/vulkan/${VULKAN_SDK_VERSION}/x86_64/include/* /usr/local/include/ \
    && mkdir -p /usr/local/lib && cp -ra /opt/vulkan/${VULKAN_SDK_VERSION}/x86_64/lib/* /usr/local/lib/ \
    && cp -a /opt/vulkan/${VULKAN_SDK_VERSION}/x86_64/lib/libVkLayer_*.so /usr/local/lib \
    && mkdir -p /usr/local/share/vulkan/explicit_layer.d \
    && cp /opt/vulkan/${VULKAN_SDK_VERSION}/x86_64/share/vulkan/explicit_layer.d/VkLayer_*.json /usr/local/share/vulkan/explicit_layer.d \
    && mkdir -p /usr/local/share/vulkan/registry \
    && cp -a /opt/vulkan/${VULKAN_SDK_VERSION}/x86_64/share/vulkan/registry/* /usr/local/share/vulkan/registry \
    && cp -a /opt/vulkan/${VULKAN_SDK_VERSION}/x86_64/bin/* /usr/local/bin \
    && ldconfig \
    && rm /tmp/vulkansdk-linux-x86_64-${VULKAN_SDK_VERSION}.tar.xz && rm -rf /opt/vulkan

#505