NVlabs/curobo #417: Mesh Face Count Mismatch in Isaac Sim

Hermanye996, 2024-10-24:
- **Isaac Sim Version**: 4.2.0
- **Python Version**: 3.10.14

#### Description
When updating the environment obstacles using assets from Isaac Sim, there are warnings indicating that the number of mesh faces does not match the expected face vertex counts. 

#### Warning Log
```bash
2024-10-24 02:48:53 [65,584ms] [Warning] [curobo] Mesh faces 1322.6666666666667 are not matching faceVertexCounts 1024
2024-10-24 02:48:53 [65,586ms] [Warning] [curobo] Mesh faces 2311.3333333333335 are not matching faceVertexCounts 1734
2024-10-24 02:48:53 [65,588ms] [Warning] [curobo] Mesh faces 1830.0 are not matching faceVertexCounts 1374
2024-10-24 02:48:53 [65,589ms] [Warning] [curobo] Mesh faces 293.3333333333333 are not matching faceVertexCounts 220
2024-10-24 02:48:53 [65,783ms] [Warning] [curobo] Mesh faces 201514.66666666666 are not matching faceVertexCounts 151136
2024-10-24 02:48:53 [65,799ms] [Warning] [curobo] Mesh faces 1702.6666666666667 are not matching faceVertexCounts 1276
2024-10-24 02:48:53 [65,801ms] [Warning] [curobo] Mesh faces 1702.6666666666667 are not matching faceVertexCounts 1276
2024-10-24 02:48:53 [65,802ms] [Warning] [curobo] Creating new Mesh cache: 52
```

#### Method Used
The issue arises when using the following method to update the obstacle world from Isaac Sim:

```python
def update_obstacle_from_isaac_sim(
    self, world, robot_prim_path: str, ignore_prim_paths: List[str] = None
) -> WorldConfig:
    """Update obstacle world from Isaac Sim.

    Args:
        world (World): The world object from Isaac Sim.
        robot_prim_path (str): The root prim path of the robot.
        ignore_prim_paths (List[str]): The prim paths of the obstacles to ignore.

    Returns:
        WorldConfig: The updated obstacle world.
    """
    # Init curobo usd helper
    if self.curobo_usd_helper is None:
        self.curobo_usd_helper = UsdHelper()
    if self.curobo_usd_helper.stage is None:
        self.curobo_usd_helper.load_stage(world.stage)

    all_items = world.stage.Traverse()
    for item in all_items:
        for key_to_ignore in ignore_prim_paths:
            if key_to_ignore in str(item.GetPath()):
                self.logger.log_debug(f"Ignoring: {str(item.GetPath())}")

    # Get obstacles (mesh, cube,...) from Isaac Sim
    obstacles = self.curobo_usd_helper.get_obstacles_from_stage(
        reference_prim_path=robot_prim_path,
        ignore_substring=ignore_prim_paths,
    ).get_collision_check_world()
    
    # Update curobo obstacle world
    self.logger.log_debug("Updating obstacles from Isaac Sim...")
    if self.ik_solver is not None:
        self.ik_solver.update_world(obstacles)
    if self.motion_gen is not None:
        self.motion_gen.update_world(obstacles)
    
    self.logger.log_debug(f"Updated {len(obstacles.objects)} obstacles.")
    self.logger.log_debug(f"mesh: {len(obstacles.mesh)}")
    return obstacles
```

Hermanye996, 2024-10-24:
Curobo seems to only support triangular mesh formats and does not support quadrilateral faces.

balakumar-s, 2026-05-18:
We added support for quad faces in https://github.com/NVlabs/curobo/commit/1c6f3a0742c8b525da3c58cb5307d009310c4bcf