NVlabs/curobo #418: Problem with motion_gen_reacher.py

Sui0206, 2024-10-24:
1. cuRobo installation mode (choose from [python, isaac sim, docker python, docker isaac sim]):  isaac sim
2. python version: 3.10
3. Isaac Sim version (if using): 4.0.0

Hi, I want to run curobo with Isaac Sim, and when I tried to run motion_gen_reacher.py , it can't work as tutorial. When I add object, the arm can't generate motion in a collision-free manner.

And it gives a warning:
2024-10-24 08:02:03 [302,154ms] [Warning] [curobo] Object already in warp cache, using existing instance for: /World/obstacles/table_mesh

Treeeplanter, 2025-03-07:
same question, have you fixed it?

Treeeplanter, 2025-03-22:
Hi, I figured that after drag the icon of obstacles from the props, we need to manually move the obstacle to /world, otherwise it will be fixed on /robot, then the robot will trate the obstacle as part of world and do collision freee plan, hope this can help!

Yandong-Luo, 2025-03-25:
> Hi, I figured that after drag the icon of obstacles from the props, we need to manually move the obstacle to /world, otherwise it will be fixed on /robot, then the robot will trate the obstacle as part of world and do collision freee plan, hope this can help!

But I didn't setup any obstacles. The manipulation also couldn't not reach the expected position. Did you have the same situation? @Treeeplanter 

[image]

my-nus, 2025-04-17:
> Hi, I figured that after drag the icon of obstacles from the props, we need to manually move the obstacle to /world, otherwise it will be fixed on /robot, then the robot will trate the obstacle as part of world and do collision freee plan, hope this can help!

I solved the problem. Thank you.

closed 2026-05-19 by balakumar-s