NVlabs/curobo #434: Joint names not taken into account when using `plan_single_js` (`MotionGen`)

mbusy, 2024-11-19:
## Environment
1. cuRobo installation mode (choose from [python, isaac sim, docker python, docker isaac sim]): `python`
2. python version: `3.8.10`
3. Isaac Sim version (if using): - 

## Issue Details
It seems that joint names in the `JointState` start and goal objects are not taken into account when using `plan_single_js` (`MotionGen`).

* I am first creating a `MotionGen` object, and give it a `MotionGenConfig`. My robot is described by that config (for the sake of this example, let's say that it's a 3dof robot with the following joint names: [`joint_1`, `joint_2`, `joint_3`])

* I then want to get a plan using `plan_single_js`. I create `JointState` objects for the start and goal states. For each, I specify the `position` and `joint_names` components.

* It seems that defining the joint names in a different order, eg [`joint_1`, `joint_3`, `joint_2`], or specifying something incorrect eg [`test`, `non_existing_joint`] won't impact the trajectory generation in any way. It looks like the planner uses the positions specified in the joint states but discards the `joint_names` component, and uses the joint_names order defined in the `MotionGenConfig` instead.

Is this normal? Can a user specify starts and goals with `JointState` objects and order the positions in any way they want, as long as the `joint_names` component is specified? 

A snippet to illustrate what I mean:
```python
# Motion gen config contains my robot config, with all its joint names defined ["joint_1", "joint_2", "joint_3"]
motion_gen = MotionGen(config=motion_gen_config)
motion_gen.warmup(enable_graph=True)

# Creating the start joint state, with positions for joint_1, joint_3, joint_2
start_joint_state = JointState.from_position(
    torch.FloatTensor(start_positions).unsqueeze(0).to(tensor_args.device),
    joint_names=["joint_1", "joint_3", "joint_2"],
)

# Creating the goal joint state, with positions for joint_1, joint_3, joint_2
goal_joint_state = JointState.from_position(
    torch.FloatTensor(goal_positions).unsqueeze(0).to(tensor_args.device),
    joint_names=["joint_1", "joint_3", "joint_2"],
)

result = motion_gen.plan_single_js(
    start_state=start_joint_state,
    goal_state=goal_joint_state,
    plan_config=MotionGenPlanConfig(max_attempts=1),
)

# I can repeat the same process keeping the same start_positions and goal_positions but use different
# joint_names, the generated trajectory will be the same
```

balakumar-s, 2024-11-19:
We currently don't check joint names and assume they are in the correct order. However, you can use this function to reorder: https://curobo.org/_api/curobo.wrap.reacher.motion_gen.html#curobo.wrap.reacher.motion_gen.MotionGen.get_active_js

mbusy, 2024-11-21:
@balakumar-s Thanks for the info, I'll reorder :+1: I'm completely OK to close the issue, or should we can keep it open if you think this joint name checking is a relevant addition to the stack?

balakumar-s, 2025-01-04:
Let's close this issue.

closed 2025-01-04 by balakumar-s