NVlabs/curobo #447: PoseCostMetric `offset_position` parameter doesn't respect negative values

KySmith1, 2024-12-19:
If it’s not a bug, please use discussions: https://github.com/NVlabs/curobo/discussions

Please provide the below information in addition to your issue:

1. cuRobo installation mode (choose from [python, isaac sim, docker python, docker isaac sim]):
Docker ([nvcr.io/nvidia/pytorch:23.12-py3](https://docs.nvidia.com/deeplearning/frameworks/pytorch-release-notes/rel-23-12.html)]
2. python version:
Python 3.10
3. Isaac Sim version (if using): 
N/A (Not in use)

------
Issue Details

When specified, pose cost metric offset positions are not accepting negative values. As a result, the resultant motions are always in the "positive" direction with respect to the URDF coordinate frame. For example:

```
    plan_config.pose_cost_metric = PoseCostMetric(
        hold_partial_pose=True,
        hold_vec_weight=tensor_args.to_device([1.0, 1.0, 1.0, 1.0, 1.0, 0.0]),
        offset_position=tensor_args.to_device([0.0, 0.0, 0.4]),
        offset_tstep_fraction=0.55,
    )
```
[Screencast from 2024-12-19 13-13-28.webm](https://github.com/user-attachments/assets/efed7281-d7ce-4cc5-961a-9f18548f610f)

This is expected, but the result is the same if the following is provided:

```
    plan_config.pose_cost_metric = PoseCostMetric(
        hold_partial_pose=True,
        hold_vec_weight=tensor_args.to_device([1.0, 1.0, 1.0, 1.0, 1.0, 0.0]),
        offset_position=tensor_args.to_device([0.0, 0.0, -0.4]),
        offset_tstep_fraction=0.55,
    )
```

It would be expected for the final pose to be below the partial pose on the Z axis. At the moment, it doesn't appear to be possible to have the final pose be below the partial pose with the current API.

balakumar-s, 2025-02-13:
The offset position is an offset from the target pose. 

Assuming the pose cost is in the world frame and world z is up.

When the offset_position is positive:
The robot will reach a point above the target pose and then linearly move down to the target pose

When the offset_position is negative:
The robot will reach a point below the target pose and then linearly move up to the target pose

In your video, the effect seems opposite. 

The implementation should work for positive and negative offsets as seen here: https://github.com/NVlabs/curobo/blob/2fbffc35225398cf9d5f382804faa9de2608753b/src/curobo/curobolib/cpp/pose_distance_kernel.cu#L227

A question is whether you have this flag True or False: https://github.com/NVlabs/curobo/blob/2fbffc35225398cf9d5f382804faa9de2608753b/src/curobo/wrap/reacher/motion_gen.py#L241

balakumar-s, 2025-04-25:
The offset pose cost has some major gaps. We are thinking of deprecating this feature and instead ask users to use MotionGen.plan_grasp() instead as that's more robust. Closing this issue as we are going to be removing this feature in the next release. Re-open if you think this feature is a must have compared to using plan_grasp instead.

closed 2025-04-25 by balakumar-s