NVlabs/curobo #449: MotionGen Status.FINETUNE TRAJOPT FAIL on Jetson AGX Orin

Shimadawallace, 2024-12-27:
1. cuRobo installation mode: python 
2. python version: python 3.10

------
Issue Details
velocity_scale set to 1.0

Hello,
I found that when I deploy my code on Jetson AGX Orin, the MotionGen.plan_single() always got no solution with "MotionGen Status.FINETUNE TRAJOPT FAIL" but it works well on X86 isaac_sim with same configuration and urdf. Could you give me some advice on how to solve the problem?

So far i tried to set  enable_finetune_trajopt=False but it still got no solution 9 out of 10 times on Jetson AGX Orin.

MotionGen configuration (isaac_sim):
`
    world_cfg = WorldConfig(cuboid=world_cfg_table.cuboid, mesh=world_cfg1.mesh)

    n_obstacle_cuboids = 30
    n_obstacle_mesh = None
    trajopt_dt = None
    optimize_dt = True
    trajopt_tsteps = 32
    trim_steps = None
    max_attempts = 4
    interpolation_dt = 0.08333
    if args.reactive:
        trajopt_tsteps = 40
        trajopt_dt = 0.04
        optimize_dt = False
        max_attempts = 1
        trim_steps = [1, None]
        interpolation_dt = trajopt_dt
    motion_gen_config = MotionGenConfig.load_from_robot_config(
        robot_cfg,
        world_cfg,
        tensor_args,
        collision_checker_type=CollisionCheckerType.MESH,
        num_trajopt_seeds=12,
        num_graph_seeds=12,
        interpolation_dt=interpolation_dt,
        collision_cache={"obb": n_obstacle_cuboids, "mesh": n_obstacle_mesh},
        optimize_dt=optimize_dt,
        trajopt_dt=trajopt_dt,
        trajopt_tsteps=trajopt_tsteps,
        trim_steps=trim_steps,
    )
    motion_gen = MotionGen(motion_gen_config)
    print("warming up...")
    motion_gen.warmup(enable_graph=True, warmup_js_trajopt=False, parallel_finetune=True)
    print("Curobo is Ready")

    add_extensions(simulation_app, args.headless_mode)

    plan_config = MotionGenPlanConfig(
        enable_graph=False,
        enable_graph_attempt=2,
        max_attempts=max_attempts,
        enable_finetune_trajopt=False,
        parallel_finetune=True,
        time_dilation_factor=0.5,
    )

`
MotionGen configuration (Jetson AGX orin)
`
        motion_gen_config = MotionGenConfig.load_from_robot_config(
            self.robot_cfg,
            world_cfg,
            self.tensor_args,
            collision_checker_type=CollisionCheckerType.MESH,
            num_trajopt_seeds=12,
            num_graph_seeds=12,
            interpolation_dt=interpolation_dt,
            collision_cache={"obb": n_obstacle_cuboids, "mesh": n_obstacle_mesh},
            optimize_dt=optimize_dt,
            trajopt_dt=trajopt_dt,
            trajopt_tsteps=trajopt_tsteps,
            trim_steps=trim_steps,
        )
        self.motion_gen = MotionGen(motion_gen_config)
        print("stable motion generator warming up...")
        self.motion_gen.warmup(enable_graph=True, warmup_js_trajopt=False, parallel_finetune=True)
        print("stable motion generator is Ready")

        self.plan_config = MotionGenPlanConfig(
            enable_graph=False,
            enable_graph_attempt=2,
            max_attempts=max_attempts,
            enable_finetune_trajopt=False,
            parallel_finetune=True,
            time_dilation_factor=0.5,
        )
`
The velocity_scale is set to 1.0

The environment on Jetson AGX orin:
[image]

balakumar-s, 2025-01-04:
Can you check  the error in motion_gen_result.position_error? This seems like the simulation gains are not tuned correctly for your robot. Can you increase the stiffness?

Shimadawallace, 2025-01-06:
Thanks for your kind reply!
the printout of "motion_gen_result.position_error" = tensor(0.0001, device='cuda:0')
Increasing the stiffness by 100 times fix the problem.
However, when I test it on real robot, there's a 6 cm position error on z axis and the the printout of "motion_gen.result.position_error" =  tensor(2.7963e-05, device='cuda:0'). Is it possibly caused by not tuned gains?

Shimadawallace, 2025-01-06:
I fixed the position error by adjusting urdf. Thanks!

closed 2025-01-06 by Shimadawallace