NVlabs/curobo #450: Plan_single_js Generate Collision Trajectory

Alice-ly, 2024-12-31:
Hi, I have been using curobo for months and it performs quite well when there is no obstacle in the world, but when I add an obstacle in the world(for example a box from Rviz scene objects), **curobo will generate trajectory that does not avoid the obstacle totally**. 

And I have tried several parameters to fix the bug (shown below), some parameters do make a difference, but it still can't avoid obstacles in all cases. 
This is the function I use :
[image]

These are parameters I have tried to change:
particle_trajopt.yml
[image]
gradient_trajopt.yml
[image]

And I feel that **only activation distance changed the trajectory, when activation distance raises, the trajectory  will avoid the obstacles but sometimes it still move through the obstacle directly**.
Could you please offer some more ideas on how to fix this problem?  thank you very much !

------
Issue Details

balakumar-s, 2025-01-04:
Are you using it though ROS or directly in python? If through python, can you provide a minimal example of the issue?

If through ROS, can you ask your question here: https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_cumotion/issues

balakumar-s, 2025-02-13:
Please re-open if the issue still exists.

closed 2025-02-13 by balakumar-s