NVlabs/curobo #481: Run curobo on H100

KIKI007, 2025-03-21:
If it’s not a bug, please use discussions: https://github.com/NVlabs/curobo/discussions

Please provide the below information in addition to your issue:

1. cuRobo installation mode (choose from [python, isaac sim, docker python, docker isaac sim]): python
2. python version: 3.11
3. Isaac Sim version (if using): 

------
Issue Details

I would like to run curobo on Nvidia H100.  

1. The pip compilation works well.

2. ```python examples/motion_gen_example.py``` fails, leading to the following errors:

``` bash
Warp CUDA error 715: an illegal instruction was encountered (in function cuda_stream_unregister, /builds/omniverse/warp/warp/native/warp.cu:2397)
Warp CUDA error 715: an illegal instruction was encountered (in function cuda_stream_destroy, /builds/omniverse/warp/warp/native/warp.cu:2371)
Warp CUDA error 715: an illegal instruction was encountered (in function cuda_unload_module, /builds/omniverse/warp/warp/native/warp.cu:3332)
Warp CUDA error 715: an illegal instruction was encountered (in function cuda_unload_module, /builds/omniverse/warp/warp/native/warp.cu:3332)
Warp CUDA error 715: an illegal instruction was encountered (in function cuda_unload_module, /builds/omniverse/warp/warp/native/warp.cu:3332)
```

3.  ```python examples/kinematics_example.py ``` works.

4. all pytest examples are failed.

balakumar-s, 2025-03-22:
Do you have a recent CUDA version installed in the python environment? What version of warp and CUDA do you have?

KIKI007, 2025-03-23:
Thank you very much for your reply.

I tried pytorch 2.8.0 + cuda12.8, which works
I tried pytorch 2.4.0 + cuda12.4, which still has the same errors. 

The warp version is 1.6.2

KIKI007, 2025-03-23:
some extra data points

pytorch 2.6.0 + cuda 12.6 works

pytorch 2.6.0 + cuda 11.8 does not work

Seems that the Cuda version cannot be lower than 12.6

balakumar-s, 2025-04-25:
This would be because of H100 requiring CUDA>=12.6.

closed 2025-04-25 by balakumar-s