NVlabs/curobo #486: robot can't grape the object

Yandong-Luo, 2025-03-23:
If it’s not a bug, please use discussions: https://github.com/NVlabs/curobo/discussions

Please provide the below information in addition to your issue:

1. cuRobo installation mode (choose from [python, isaac sim, docker python, docker isaac sim]):python
2. python version: 3.10
3. Isaac Sim version (if using): 4.2.0

------
Issue Details

when I run `omni_python examples/isaac_sim/batch_motion_gen_reacher.py`, the robot can't grape the object

[image]

When I run `omni_python examples/isaac_sim/motion_gen_reacher.py`, I meet the same issue

[image]

balakumar-s, 2025-04-25:
What do you see in the terminal?

balakumar-s, 2025-04-25:
Can you try the latest commit? We increased the velocity threshold: https://github.com/NVlabs/curobo/blob/0a50de1ba72db304195d59d9d0b1ed269696047f/examples/isaac_sim/motion_gen_reacher.py#L365

Yandong-Luo, 2025-04-25:
[image]

It's same. That's so weird. But multi_arm_reacher.py is correct.

[image]

balakumar-s, 2025-04-25:
Did you try moving the red cube to a few different positions?

Yandong-Luo, 2025-04-25:
yep, it's the same.

[image]

balakumar-s, 2025-04-25:
Is it possible to use Isaac Sim 4.5? [I am not sure if the Isaac Sim version is the problem, but we just tested this example on Isaac Sim 4.5, so it should definitely work]

Yandong-Luo, 2025-04-25:
I also use sim 4.5. It's the same problem. When I run `omni_python -m pytest .`, I got:
19:36:44 in curobo on  main [?] 
➜  omni_python -m pytest .                             
=============================================================================================================================== test session starts ===============================================================================================================================
platform linux -- Python 3.10.15, pytest-8.3.5, pluggy-1.5.0
rootdir: /home/chris/curobo
configfile: pyproject.toml
plugins: anyio-4.6.2.post1
collected 264 items / 4 skipped                                                                                                                                                                                                                                                   

tests/cost_test.py .                                                                                                                                                                                                                                                        [  0%]
tests/cuda_graph_test.py .                                                                                                                                                                                                                                                  [  0%]
tests/cuda_robot_generator_test.py ....                                                                                                                                                                                                                                     [  2%]
tests/curobo_robot_world_model_test.py ............                                                                                                                                                                                                                         [  6%]
tests/curobo_version_test.py .                                                                                                                                                                                                                                              [  7%]
tests/geom_test.py ......                                                                                                                                                                                                                                                   [  9%]
tests/geom_types_test.py ......                                                                                                                                                                                                                                             [ 11%]
tests/goal_test.py ..                                                                                                                                                                                                                                                       [ 12%]
tests/ik_config_test.py ....                                                                                                                                                                                                                                                [ 14%]
tests/ik_module_test.py ......                                                                                                                                                                                                                                              [ 16%]
tests/ik_test.py .....                                                                                                                                                                                                                                                      [ 18%]
tests/interpolation_test.py .                                                                                                                                                                                                                                               [ 18%]
tests/kinematics_test.py .......                                                                                                                                                                                                                                            [ 21%]
tests/mimic_joint_test.py ..                                                                                                                                                                                                                                                [ 21%]
tests/motion_gen_api_test.py ...                                                                                                                                                                                                                                            [ 23%]
tests/motion_gen_constrained_test.py ............................                                                                                                                                                                                                           [ 33%]
tests/motion_gen_cuda_graph_test.py .............                                                                                                                                                                                   [... cut ...] ollada-0.9 referencing-0.36.2 rpds-py-0.24.0 rtree-1.4.0 scikit-image-0.25.2 scipy-1.15.2 shapely-2.1.0 svg.path-6.3 tifffile-2025.3.30 tqdm-4.67.1 trimesh-4.6.8 triton-3.1.0 vhacdx-0.0.8.post2 warp-lang-1.7.0 xxhash-3.5.0 yourdfpy-0.0.57

[notice] A new release of pip is available: 24.0+nv1 -> 25.0.1
[notice] To update, run: /home/chris/isaacsim-4.5.0/kit/python/bin/python3 -m pip install --upgrade pip

I am not sure whether this warning and error will effect the result of motion_gen_reacher.py