NVlabs/curobo #492: ERROR for ik_config_test and trajopt_config_test

Treeeplanter, 2025-04-07:
If it’s not a bug, please use discussions: https://github.com/NVlabs/curobo/discussions

Please provide the below information in addition to your issue:

1. cuRobo installation mode (choose from [python, isaac sim, docker python, docker isaac sim]): python 
2. python version: 3.10.16
3. Isaac Sim version (if using): 

------
Issue Details
 hi, I'm trying to install curobo as a package in python, but can not pass the test after installition. 

My Env:
PyTorch version: 2.1.0+cu118
CUDA available: True
CUDA version (from torch): 11.8
Device count: 1
Device name: NVIDIA GeForce RTX 4090

I have followed the installiztion torturial in the website of curobo and get this:

# pip show nvidia-curobo
Name: nvidia_curobo
Version: 0.7.6
Summary: GPU robot motion toolkit containing cuda accelerated kinematics, IK, MPC, Global motion planning, and optimization solvers.
Home-page: https://curobo.org
Author: NVIDIA Seattle Robotics Lab
Author-email: 
License: NVIDIA
Location: /home/shuo/anaconda3/envs/mjthor/lib/python3.10/site-packages
Editable project location: /home/shuo/research/mujoco-thor/external_src/curobo
Requires: importlib_resources, networkx, numpy, numpy-quaternion, pybind11, pyyaml, scikit-image, scipy, setuptools_scm, torch, tqdm, trimesh, warp-lang, wheel, yourdfpy

then I run test and found that I can not pass the test:

# python3 -m pytest .
========================================================================= test session starts =========================================================================
platform linux -- Python 3.10.16, pytest-8.3.5, pluggy-1.5.0
rootdir: /home/shuo/research/mujoco-thor/external_src/curobo
configfile: pyproject.toml
plugins: anyio-4.9.0, dash-3.0.2
collected 258 items / 2 errors / 3 skipped                                                                                                                            

=============================================================================== ERRORS ================================================================================
______________________________________________________________ ERROR collecting tests/ik_config_test.py _______________________________________________________________
tests/ik_config_test.py:134: in <module>
    (ik_base_config(), True),
tests/ik_config_test.py:39: in ik_base_config
    ik_config = IKSolverConfig.load_from_robot_config(
/home/shuo/anaconda3/envs/mjthor/lib/python3.10/contextlib.py:79: in inner
    return func(*args, **kwds)
src/curobo/wrap/reacher/ik_solver.py:293: in load_from_robot_config
    grad_cfg = ArmReacherConfig.from_dict(
/home/shuo/anaconda3/envs/mjthor/lib/python3.10/contextlib.py:79: in inner
    return func(*args, **kwds)
src/curobo/rollout/arm_base.py:195: in from_dict
    cost = cls.cost_from_dict(
src/curobo/rollout/arm_reacher.py:141: in cost_from_dict
    return ArmReacherCostConfig.from_dict(
src/curobo/rollout/arm_reacher.py:118: in from_dict
    data = ArmCostConfig._get_formatted_dict(
src/curobo/rollout/arm_base.py:94: in _get_formatted_dict
    data[k] = cost_key_list[k](**data_dict[k], tensor_args=tensor_args)
<string>:19: in __init__
    ???
src/curobo/rollout/cost/dist_cost.py:44: in __post_init__
    return super().__post_init__()
src/curobo/rollout/cost/cost_base.py:39: in __post_init__
    self.weight = self.tensor_args.to_device(self.weight)
src/curobo/types/base.py:35: in to_device
    return torch.as_tensor(np.array(data_tensor), device=self.device, dtype=self.dtype)
E   TypeError: len() of unsized object
____________________________________________________________ ERROR collecting tests/trajopt_config_test.py ____________________________________________________________
tests/trajopt_config_test.py:115: in <module>
    (trajopt_base_config(), True),
tests/trajopt_config_test.py:35: in trajopt_base_config
    trajopt_config = TrajOptSolverConfig.load_from_robot_config(
/home/shuo/anaconda3/envs/mjthor/lib/python3.10/contextlib.py:79: in inner
    return func(*args, **kwds)
src/curobo/wrap/reacher/trajopt.py:399: in load_from_robot_config
    grad_cfg = ArmReacherConfig.from_dict(
/home/shuo/anaconda3/envs/mjthor/lib/python3.10/contextlib.py:79: in inner
    return func(*args, **kwds)
src/curobo/rollout/arm_base.py:195: in from_dict
    cost = cls.cost_from_dict(
src/curobo/rollout/arm_reacher.py:141: in cost_from_dict
    return ArmReacherCostConfig.from_dict(
src/curobo/rollout/arm_reacher.py:118: in from_dict
    data = ArmCostConfig._get_formatted_dict(
src/curobo/rollout/arm_base.py:94: in _get_formatted_dict
    data[k] = cost_key_list[k](**data_dict[k], tensor_args=tensor_args)
<string>:19: in __init__
    ???
src/curobo/rollout/cost/dist_cost.py:44: in __post_init__
    return super().__post_init__()
src/curobo/rollout/cost/cost_base.py:39: in __post_init__
    self.weight = self.tensor_args.to_device(self.weight)
src/curobo/types/base.py:35: in to_device
    return torch.as_tensor(np.array(data_tensor), device=self.device, dtype=self.dtype)
E   TypeError: len() of unsized object
======================================================================= short test summary info =======================================================================
ERROR tests/ik_config_test.py - TypeError: len() of unsized object
ERROR tests/trajopt_config_test.py - TypeError: len() of unsized object
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! Interrupted: 2 errors during collection !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
==================================================================== 3 skipped, 2 errors in 2.16s =====================================================================. It did not happen for another repo, where I also used curobo as another source, can you help me? thank you!

balakumar-s, 2025-04-25:
Not sure what's happening here. We do have it installed and working with python 3.10. Are you able to run this inside a docker on the same machine?

Treeeplanter, 2025-04-25:
Thank you!
I have solved this problem by changing the sequence of install curobo and
ompl, I don't know why.
I found that I should install curobo first then the version will be 0.0.0,
otherwise it will be 1.7.3.