NVlabs/curobo #499: .INVALID_START_STATE_JOINT_LIMITS and  MotionGenStatus.IK_FAIL

gaolin131, 2025-04-29:
I have a humanoid robot. When using curobo for motion planning, I frequently encounter error messages like "Plan did not converge to a solution: MotionGenStatus.INVALID_START_STATE_JOINT_LIMITS" and "Plan did not converge to a solution: MotionGenStatus.IK_FAIL". However, when I use other methods such as reinforcement learning or MoveIt, these issues do not occur. I've ensured that the URDF remains consistent across all methods. Currently, I suspect that the problem might be caused by the structural fitting between links. Do you have any good suggestions for this situation? Are there any similar cases?

balakumar-s, 2025-05-14:
For  MotionGenStatus.INVALID_START_STATE_JOINT_LIMITS, it might be because of the start state being outside of joint limits in the urdf. Can you share the urdf and the curobo robot configuration file? A minimal script to reproduce this error will also help.

Dwl2021, 2025-05-15:
I've also encountered this issue with `MotionGenStatus.INVALID_START_STATE_JOINT_LIMITS`. I had a joint limit of `0-pi`, and the homestate was `0`, but when imported into the simulator it produced a very tiny error, such as `-1e-8`. This situation causes the start_state to be invalid. The best solution is to use `round` before planning to avoid floating point error, which has almost no effect on performance. Alternatively, you could set the homestate to be less extreme. Hope this helps you.🙂