NVlabs/curobo #502: Shape Mismatch Error in `plan_batch`

Dwl2021, 2025-05-03:
If it’s not a bug, please use discussions: https://github.com/NVlabs/curobo/discussions

Please provide the below information in addition to your issue:

1. cuRobo installation mode: python
2. python version: Python 3.10.0
3. Isaac Sim version (if using): not using

------

## Description
I encountered a `RuntimeError` due to a shape mismatch when using `plan_batch` in curobo.

The error screenshot is presented below:

[image]

## Analysis
I tried to find the root cause of this problem, so I looked through the file named `motion_gen.py`. And I found that in line 3412, the `result.success` is set to all false as follows:
```python
result.success[:] = False
if not graph_result.valid_query:
    result.valid_query = False
    if self.store_debug_in_result:
        result.debug_info["graph_debug"] = graph_result.debug_info
    return result
if plan_config.need_graph_success:
    return result
```
However, it did not return immediately but continued to run to line 3426 
```python
self._trajopt_goal_config[:, :ik_success] = goal_config
```
and then output an error of mismatch.

balakumar-s, 2025-05-14:
Can you provide a minimal script to reproduce this error?

Dwl2021, 2025-05-15:
The workspace has been compressed and uploaded. You need to modify the absolute path manually.

[scripts.zip](https://github.com/user-attachments/files/20219726/scripts.zip)

Main script as follows,
```
# Third Party
import torch

# cuRobo
from curobo.types.math import Pose
from curobo.types.robot import JointState
from curobo.wrap.reacher.motion_gen import MotionGen, MotionGenConfig, MotionGenPlanConfig

world_config = {
    "cuboid": {
        "table": {
            "dims": [5.0, 5.0, 0.2],  # x, y, z
            "pose": [100.0, 0.0, -0.1, 1, 0, 0, 0.0],  # x, y, z, qw, qx, qy, qz
        },
    },
}

motion_gen_config = MotionGenConfig.load_from_robot_config(
    "/home/algo/dengweiliang/curobo_ws/scripts/curobo.yml",
    world_config,
    interpolation_dt=1/250,
    num_trajopt_seeds=1,
    num_graph_seeds=1
)
motion_gen = MotionGen(motion_gen_config)
motion_gen.warmup(batch=10)

goal_pose = Pose.from_list([0.0, -0.0, 1e-05, 0.0, -0.0, 0.0])  # x, y, z, qw, qx, qy, qz
tensor = torch.tensor([[ 0.5067, -0.3133,  0.1340,  0.2119,  0.2726,  0.6549, -0.6723],
        [ 0.5078, -0.2914,  0.1335,  0.2443,  0.2387,  0.6435, -0.6851],
        [ 0.5073, -0.2695,  0.1315,  0.2762,  0.2041,  0.6304, -0.6961],
        [ 0.5050, -0.2479,  0.1281,  0.3073,  0.1691,  0.6159, -0.7055],
        [ 0.5011, -0.2268,  0.1234,  0.3377,  0.1336,  0.5997, -0.7130],
        [ 0.4956, -0.2063,  0.1174,  0.3673,  0.0978,  0.5821, -0.7188],
        [ 0.4885, -0.1868,  0.1101,  0.3959,  0.0618,  0.5630, -0.7228],
        [ 0.4799, -0.1684,  0.1016,  0.4236,  0.0256,  0.5425, -0.7250],
        [ 0.4700, -0.1513,  0.0921,  0.4502, -0.0107,  0.5207, -0.7254],
        [ 0.4587, -0.1357,  0.0815,  0.4756, -0.0469,  0.4975, -0.7239]])
tensor = tensor.cuda()
goal_pose_of_gripper = Pose(tensor[:, :3], tensor[:, 3:])
joint_angles = [0.0, -0.0, 1e-05, 0.0, -0.0, 0.0]
joint_angles_cuda = torch.tensor(joint_angles, dtype=torch.float32).cuda().reshape(1, -1)
joint_angles_cuda = torch.cat([joint_angles_cuda] * 10, dim=0)
start_joint_states = JointState.from_position(
    joint_angles_cuda,
    joint_names=['fl_joint1', 'fl_joint2', 'fl_joint3', 'fl_joint4', 'fl_joint5', 'fl_joint6']
)
plan_config = MotionGenPlanConfig(max_attempts=10)
result = motion_gen.plan_batch(
                start_joint_states, 
                goal_pose_of_gripper, 
                plan_config
            )
```