NVlabs/curobo #503: The same goal pose cannot be solved the second time, but solvable when resetting the cuda_graph

Ericonaldo, 2025-05-08:
If it’s not a bug, please use discussions: https://github.com/NVlabs/curobo/discussions

Please provide the below information in addition to your issue:

1. cuRobo installation mode: [python]):
2. python version: 3.10.16
3. Isaac Sim version (if using): Not using

------
Issue Details

We found that at some specific points, the solver can only succeed when resetting the cuda_graph. Therefore, we find out the goal pose at the failed time, and construct the smallest debug case as below. As you can see, we solve the same goal pose twice, and we failed at the second time, which can be success if we reset the cuda_graph. 

```
# cuRobo
from curobo.types.base import TensorDeviceType
from curobo.types.math import Pose
from curobo.types.robot import RobotConfig
from curobo.cuda_robot_model.cuda_robot_model import CudaRobotModel
from curobo.wrap.reacher.ik_solver import IKSolver, IKSolverConfig
from curobo.util_file import load_yaml
import torch
urdf_path = "robot.urdf" #UR-5 arm

tensor_args = TensorDeviceType(device=torch.device("cuda"))
robot_cfg = RobotConfig.from_basic(
            urdf_path, "base_link", "wrist_3_link", tensor_args
        )
ik_config = IKSolverConfig.load_from_robot_config(
            robot_cfg,
            None,
            position_threshold=0.005,
            num_seeds=1,
            self_collision_check=True,
            self_collision_opt=True,
            tensor_args=tensor_args,
            high_precision=False,
            use_cuda_graph=True,
            grad_iters=None,
            regularization=True,
        )
kin_model = CudaRobotModel(robot_cfg.kinematics)
ik_solver = IKSolver(ik_config)

joint_batch = torch.tensor([[-3.1393, -2.1061, -1.5349, -1.1239,  1.4910, -0.3836]], device="cuda:0").repeat(10, 1)
target_pos = torch.tensor([[ 0.6480, -0.0984,  0.2176]], device="cuda:0")
target_quat = torch.tensor([[ -0.0016,  0.8416, -0.5393,  0.0285]], device="cuda:0")

pose = Pose(target_pos, target_quat)
res = ik_solver.solve_batch(pose, retract_config=joint_batch, seed_config=joint_batch[None], return_seeds=10)
print(res.success)

ik_solver.reset_cuda_graph() # Only solvable when do this

pose = Pose(target_pos, target_quat)
res = ik_solver.solve_batch(pose, retract_config=joint_batch, seed_config=joint_batch[None], return_seeds=10)
print(res.success)
```

siangfrank, 2025-05-09:
The IK solver may succeed on the second attempt in this sequence, but I don't understand the reason:
solve target1 -> solve target2 -> solve target1.

BTW, I didn't call `warmup` for my `ik_solver`

Ericonaldo, 2025-05-09:
> The IK solver may succeed on the second attempt in this sequence, but I don't understand the reason: solve target1 -> solve target2 -> solve target1.
> 
> BTW, I didn't call `warmup` for my `ik_solver`

I am not sure what you mean. In my shown case, `target1 = target2`, and the solver fails if I do not call `ik_solver.reset_cuda_graph()` or `warm up`.

balakumar-s, 2025-05-14:
retract_config should be of shape (batch_size, dof)
seed_config should be of shape (batch_size, number_of_seeds, dof)

A few things to fix: 
1. If you want to return 10 seeds, set num_seeds to >= 10 when creating the IKSolver.
2. Fix the retract_config shape to be (1,dof)