NVlabs/curobo #505: Docker Build Fails on Vulkan SDK Download (Isaac Sim 4.0.0)

ukhyeon, 2025-05-13:
1. cuRobo installation mode : docker isaac sim
2. python version: 3.10.12
3. Isaac Sim version (if using) : 4.0.0

------
Issue Details

Hi, thank you for the great project.

I'm trying to build cuRobo with the isaac_sim_4.0.0 tag using the provided build_docker.sh script.
However, the build fails during stage 5 when trying to download and extract the Vulkan SDK:

[stage-1  5/26] RUN wget -q --show-progress ... vulkansdk-linux-x86_64-1.3.224.1.tar.gz
ERROR: failed to solve: process "/bin/sh -c wget ..." did not complete successfully: exit code: 8

From the Dockerfile:
https://sdk.lunarg.com/sdk/download/1.3.224.1/linux/vulkansdk-linux-x86_64-1.3.224.1.tar.gz
It seems this URL may no longer be accessible or the resource has been moved.
Could you update the Dockerfile with a working mirror or cached version of the Vulkan SDK?

[image]

Thanks for your help!

balakumar-s, 2025-05-14:
Does this help? https://github.com/NVlabs/curobo/issues/413

ukhyeon, 2025-05-19:
thank you for the quick response !

closed 2025-05-19 by ukhyeon